whatwg/bluetooth #442: Write characterisctic to raspberry pi fails

borch84, 2019-06-13:
Hi, I am working o a web application using web bluetooth api in order to connect a raspberry pi for writing some characteristic. 
If I write the characteristic value using nRF connect it works
[image]

But if I use the internal bluetooth interface from chrome browser it fails:
[image]

I cannot find how to get the error cause. I am able to use other bluetooth devices with my web application with no issues. 
Really appreciate if you could point me to the right direction. Thanks.

reillyeon, 2019-06-13:
Please file an issue report on bugs.chromium.org so that Chromium developers can investigate this implementation issue. Please follow [the instructions](https://www.chromium.org/developers/how-tos/file-web-bluetooth-bugs) for Android to include logs with your report.

Closing this issue as implementation-specific. Feel free to comment with a link to the Chromium issue when it has been filed but do not re-open this issue.

closed 2019-06-13 by reillyeon

borch84, 2019-06-14:
hi @reillyeon I opened the chromium issue here: https://bugs.chromium.org/p/chromium/issues/detail?id=974103
Thanks.