whatwg/bluetooth #445: does web-bluetooth support ble mesh?

zzzh, 2019-06-24:
It would be great if it does or there's a plan on it.

reillyeon, 2019-06-24:
Checking a random selection of operating systems it does not appear that either Android or Windows support Bluetooth Mesh, which means that it will be difficult for browsers to implement support for it if it were specified by Web Bluetooth.

Example code for these operating systems utilize the GATT portions of the Bluetooth Mesh specification (provisioning and proxy nodes) and these services should be accessible with Web Bluetooth without changes to the specification.

Leaving this issue open to track future support for Bluetooth Mesh's other transport layers as support begins to appear in operating systems.

zzzh, 2019-06-24:
Sounds good, thank you, reillyeon.
The issue I'm facing now is to broadcast in ADV_NONCONN_IND way,  seems Android don't support it, any ideas?

reillyeon, 2019-06-24:
Web Bluetooth does not currently support broadcasting advertisements. If you're asking how to do that with the Android SDK that is off-topic for this issue tracker.

g-ortuno, 2019-06-24:
Advertising aside, I think the rest could be implemented on top of Web Bluetooth I believe @bluetooth-mdw was looking into what was missing from the current implementation.

bluetooth-mdw, 2019-06-25:
This is a video of a Web Bluetooth application controlling a Bluetooth mesh network via a proxy node: https://youtu.be/1rxgY1EnmJw?t=1293 

I had to cut some corners though, due to limitations regarding Web Bluetooth and the degree to which it is currently implemented in browsers.

There are two interpretations of the question regarding "support for Bluetooth mesh" because there are two ways this could be approached. 

1. Using Bluetooth mesh proxy nodes
The first is support for the proxy protocol and the use of mesh proxy nodes. This involves GAP and GATT and so Web Bluetooth has *almost* all of what is needed, albeit not in the friendliest of states. To use a proxy node you must discover it, connect and then write to a characteristic and receive notifications from another characteristic, both part of the mesh proxy service. The data sent / received in those operations are proxy protocol PDUs which wrap mesh PDUs such as mesh network PDUs. 

There are several issues (that I know of):
(i) Proxy nodes identify themselves by including the mesh proxy service uuid=0x1828 in the Complete/Incomplete List of 16-bit Service UUIDs AD type in advertising packets. Web Bluetooth needs to allow filtering on this field. I believe this is specified but not implemented.
(ii) The ServiceData AD type is also included in proxy advertising packets and includes one of two possible ways of identifying a proxy node, either uniquely or by the network it belongs to (so you know the node is your network's proxy and not your neighbour's). This needs to be something you can filter on when scanning.
(iii) Formulating proxy PDUs involves creating a buffer of bits and bytes, per the specification. It's not hard but it is not exactly friendly either and it's certainly low level. You then have to secure the PDU using  functions defined in the Bluetooth mesh specification but which involve AES and AES-CCM crypto functions. There are third party APIs around which I've used (but only for demos). Ideally, if Web Bluetooth was to "support Bluetooth mesh", it would provide APIs that make the formulation and securing of proxy PDUs much easier.
(iv) Provisioning: This is the procedure followed when a new device is added to a mesh network. It results in the node being issued with a series of security keys. A Web Browser running code as described to send and receive PDUs into the mesh network, directly or via a proxy, needs to be provisioned otherwise it is not part of the network. Web Browsers are not dealt with in the mesh specification and I see some issues with the idea of provisioning a browser. I wrote a piece about this a while ago: https://www.bluetooth.com/blog/securing-mesh-proxy-applications/ In short, there's a choice of Web Bluetooth having mesh provisioning capabilities and APIs or.... this falling to the developer to create a proprietary solution for.

2. Native support for Bluetooth mesh (no proxy)

In this interpretation / approach, developers would be able to work directly with mesh models, messages, states and so on, per the Bluetooth mesh specifications and without using a mesh proxy node. This would require implementation of APIs that use what is known as the "advertising bearer". This involves unconnectable advertising packets being used as the bearer for mesh messages and scanning being used to receive them. Web Bluetooth could provide Bluetooth mesh APIs built on this bearer (and the other layers of the Bluetooth mesh stack which sit on top of the bearer layer) but only if the native platform has support for everything required. This would not be a small undertaking. It would be significantly easier if the underlying platform has comprehensive Bluetooth mesh APIs, of course.

I would recommend focusing on (1). Proxy nodes were created for this purpose anyway and on the face of it, the list of issues is not too big.

Caveat: the above represents my personal opinion and does not form an 'official' position of the Bluetooth SIG, nor is it necessarily 100% complete. The mesh specifications should be consulted for details.

An educational resource which explains and gives hands-on experience of proxy nodes and the proxy protocol, using Web Bluetooth is available from here:

https://www.bluetooth.com/bluetooth-resources/?types=study-guide&categories=&tags=mesh&keyword=

PaulLeGuennec, 2019-07-24:
Hi, I have a small question related to what @bluetooth-mdw said. 

> 1. Using Bluetooth mesh proxy nodes
The first is support for the proxy protocol and the use of mesh proxy nodes. This involves GAP and GATT and so Web Bluetooth has almost all of what is needed, albeit not in the friendliest of states. To use a proxy node you must discover it, connect and then write to a characteristic and receive notifications from another characteristic, both part of the mesh proxy service. The data sent / received in those operations are proxy protocol PDUs which wrap mesh PDUs such as mesh network PDUs.

Does it means that GATT must be implemented on a mesh proxy node to be able to read messages from a mesh network ? I have been trying to read Mesh Proxy Data Out on several applications (nrfConnect, the web bluetooth server from the mesh proxy study kit), and I did not managed to read it.

Thanks for your response.

bluetooth-mdw, 2019-07-24:
Hi, I think we're going off topic here. The best place for questions about the Bluetooth specifications themselves, is at bluetooth.com (navigate to Contact Us and you'll find Ask a Question in there).

But.... yes. However, Mesh Proxy Data Out does not support READ. It supports NOTIFY only. See 7.2.3 of the Mesh Profile specification.

Cheers

Martin

PaulLeGuennec, 2019-07-24:
Sorry, for going off topic, I'll go on bluetooth.com. And thanks for your response.

bluetooth-mdw, 2019-07-24:
No problem! I'm probably guilty of having coaxed things in that direction :-)