whatwg/bluetooth #451: getPrimaryService(s) returns nothing

dodgePT, 2019-07-31:
Hi all!

I'm experimenting with web bluetooth, trying to access services and characteristics on a BLE device, namely a smartwatch.
I'm on Win10 64bit, using electron framework but I also used the samples page to test this.
Everything goes according to plan up until I try to get services. It connects, I get the name, ID and gatt connected status, but then, when I try to get services, there's no response. It just stalls there, with no timeout, nothing.

First I thought this could be something related to electron, but then I tested all samples, and while they connect to the watch (just like my app), none of the samples are able to get any services too.

I got all UUIDs for services and characteristics of the watch, obtained through nRF Connect on Android.

Here's the simple class I made:
`class BTHuawei {
  constructor() {
    this.device = null;
    this.onDisconnected = this.onDisconnected.bind(this);
    this.characts = {};
  }

  request() {
    let options = {
      filters: [
        {
          namePrefix: "HUAWEI"
        }
      ],
      optionalServices: [
        0x1800,
        0x1801,
        0x180a,
        0x2902,
        "0000fe86-0000-1000-8000-00805f9b34fb",
        "00003802-0000-1000-8000-00805f9b34fb"
      ]
    };
    return navigator.bluetooth
      .requestDevice(options)
      .then(device => {
        this.device = device;
        this.device.addEventListener(
          "gattserverdisconnected",
          this.onDisconnected
        );
      })
      .catch(error => {
        console.log(error.message);
      });
  }

  connect() {
    console.log("Connecting...");
    if (!this.device) {
      return Promise.reject("Device is not connected.");
    }
    return this.device.gatt.connect();
  }

  //////////////////////////////////////////////////////////////////////////////

  readCharacteristic(serviceUUID, characteristicUUID) {
    return this.device.gatt
      .getPrimaryService(serviceUUID)
      .then(service => {
        service
          .getCharacteristic(characteristicUUID)
          .then(characteristic => {
            characteristic.startNotifications();
          })
          .then(characteristic => {
            characteristic.addEventListener(
              "characteristicvaluechanged",
              handleCharacteristicValueChanged
            );
            characteristic.readValue();
          });
      });
  }

  discoverServicesAndCharacteristics() {
    return this.device.gatt
      .getPrimaryServices()
      .then(services => {
        let queue = Promise.resolve();
        services.forEach(service => {
          queue = queue.then(_ =>
            service.getCharacteristics().then(characteristics => {
              log("> Service: " + service.uuid);
              characteristics.forEach(characteristic => {
                console.log(
                  ">> Characteristic: " +
                    characteristic.uuid +
                    " " +
                    getSupportedProperties(characteristic)
                );
              });
            })
          );
        });
        return queue;
      })
      .catch(error => {
        console.log(error.message);
      });
  }

  //////////////////////////////////////////////////////////////////////////////

  disconnect() {
    if (!this.device) {
      return Promise.reject("Device is not connected.");
    }
    return this.device.gatt.disconnect();
  }

  onDisconnected() {
    console.log("Device is disconnected.");
  }
}`

And this is how I use it:

`bTHuawei
	.request()
		.then(_ => bTHuawei.connect())
		.then(_ => {
			console.log(bTHuawei.device.name);
			console.log(bTHuawei.device.id);
			console.log(bTHuawei.device.gatt.connected);

			bTHuawei.readCharacteristic(0x1800, 0x2AA6);
		})
		.catch(error => {
			console.log(error.message);
		});`

Am I missing some step or something?
I guess not because none of the examples work after successfuly connecting.

Thank you for your time. Let me know if I can improve my query with more information.

dodgePT, 2019-07-31:
Just to add a little more info, I have dual boot on my production laptop, so I tried the same project on Linux Deepin x64.

On Linux, it works! What is different, I don't know, but it works.
Once gat.connected=true, I can get the services and characteristics!

reillyeon, 2019-07-31:
I am not familiar with Electron's implementation of Web Bluetooth. If you can reproduce this in Google Chrome please file an issue with this information (as well as exactly which Chrome and Windows versions you are using) on crbug.com and the Chromium developers will take a look. If not then file an issue in the Electron issue tracker.

When `getPrimaryService()` fails to resolve that likely means that the browser cannot finish the GATT service discovery process. The output from chrome://device-log with the "debug" checkbox checked may be helpful.

This issue tracker is only for Web Bluetooth specification issues and feature requests. Closing this issue.

closed 2019-07-31 by reillyeon

dodgePT, 2019-07-31:
Thank you for responding. I specified in the issue that I had tried in chrome with the same results. What's more, it works in Linux, but not in windows 10.Anyway, I chose another solution for this, thanks for your help. Enviado do meu telemóvel Huawei.-------- Mensagem original --------Assunto: Re: [WebBluetoothCG/web-bluetooth] getPrimaryService(s) returns nothing (#451)De: Reilly Grant Para: WebBluetoothCG/web-bluetooth CC: dodgePT ,Author I am not familiar with Electron's implementation of Web Bluetooth. If you can reproduce this in Google Chrome please file an issue with this information (as well as exactly which Chrome and Windows versions you are using) on crbug.com and the Chromium developers will take a look. If not then file an issue in the Electron issue tracker.
When getPrimaryService() fails to resolve that likely means that the browser cannot finish the GATT service discovery process. The output from chrome://device-log with the "debug" checkbox checked may be helpful.
This issue tracker is only for Web Bluetooth specification issues and feature requests. Closing this issue.

Jip-Hop, 2020-01-03:
@dodgePT which solution did you choose to solve this? :) I've got a problem on Windows too, but it works fine on Chrome and Electron on MacOS. On Windows it gets stuck on the "Getting Services..." step in [this example](https://googlechrome.github.io/samples/web-bluetooth/discover-services-and-characteristics.html). The device I'm trying to connect to is a [Philips Hue Smart Plug](https://www2.meethue.com/nl-nl/p/hue-smart-stekker/8718699689285).

dodgePT, 2020-01-03:
> @dodgePT which solution did you choose to solve this? :) I've got a problem on Windows too, but it works fine on Chrome and Electron on MacOS. On Windows it gets stuck on the "Getting Services..." [...]

Hi Jip-Hop!
I ended up using another framework/language (flutter). Web Bluetooth is still under development and didn't work as I intended and Electron has no stable libraries I could use for BLE, it's all a mess.
On the other hand, flutter was great and worked perfectly and it's also multi-platform. 
Sorry I can't help you more with this, maybe you should also try another approach.

Jip-Hop, 2020-01-08:
Thanks for the advice @dodgePT. Fortunately I got BLE working now with Electron and Google Chrome on Windows. The missing step was to first go to _Bluetooth & other devices_ in the Windows settings -> Add Bluetooth or other device -> _Bluetooth device_ and then choose the device you want to use for BLE. Then after these steps I could connect and request the services with Web Bluetooth.