whatwg/bluetooth #454: Pairing Screen Message:  “Turn on Bluetooth to allow pairing”

mduffy215, 2019-09-22:
Under what conditions is this message displayed?

As soon as my web app tries to pair I am getting the message, "Turn on Bluetooth to allow pairing".

On my system Bluetooth is turned on in Windows 10 and I can manually pair my workstation to my Android phone. I have fiddled with every Windows Bluetooth setting for hours now (disabling devices, etc.). 

[image]

Thx!

JamesHollyer, 2019-09-23:
The draft spec has statements about the user prompt such as:"This is intentionally vague about the details of the permission UI and how the UA infers user intent. UAs should be able to explore lots of UI within this framework." So it does not state anything about this.  If you are raising this issue in the spec because you would like some clarification on this point I am sure there could be a conversation about that and I would ask for @reillyeon 's input on that.

In the meantime this seems to be a bug in the Chrome implementation and I would like to refer you to this bug tracker: https://bugs.chromium.org/p/chromium/issues/detail?id=1006688 which should shed some light on the issue and allow you to track and contribute as we work to fix it.

reillyeon, 2019-09-24:
This seems like an entirely Chromium-side implementation issue.

closed 2019-09-24 by reillyeon