whatwg/bluetooth #459: discussion: is there possibility to change getAvailability() be sync API

yorkie, 2019-10-24:
The current `getAvailability()` is an async API and the given example looks like this:

```js
navigator.bluetooth.getAvailability().then(isAvailable => {
  // get BLE availability by the promise value
});
navigator.bluetooth.addEventListener('availabilitychanged', e => {
  // get BLE availability by `e.value`
});
```
This looks inconsistent a bit though it's working absolutely, and a little bit weird with the `async` attribute, take a look at the type of example:

```js
navigator.bluetooth.on('availabilitychanged', e => {
  navigator.bluetooth.getAvailability().then((isAvailable) => {
    assert.equal(e.value, isAvailable); // possible to be failed unfortunately
  });
});
```

Within this inconsistent but correct behavior, that's easier to produce a buggy application. See the ["Overall Bluetooth availability"](https://webbluetoothcg.github.io/web-bluetooth/#availability) section, at 2nd algorithm block, it shows up the **UA** SHOULD hold the status in memory, and fires event when it gets changed, so the case to get it still be async might be the first-time fetch? But IMO, blocking this method is another solution to us util the user type ACCEPT or REJECT.

odejesush, 2019-10-24:
Hi yorkie,

Thanks for bringing up this issue. @scheib, @reillyeon, and I had a discussion about this and concluded that it doesn't make much sense to have both APIs. Therefore, the `availabilitychanged` event will be removed from the spec. Most mobile devices and laptops have Bluetooth integrated into them. Devices such as desktop PCs that use Bluetooth adapters to enable Bluetooth functionality are probably the only ones that could really not have Bluetooth available. A site could always check for availability when it first loads and prompt the user to insert a Bluetooth adapter and reload the page.

Please let me know if you have a particular use case that needs the 'availabiltychanged' event.

thegecko, 2019-10-25:
I'm not sure I agree with removing this event. It's a common model to have functionality available to ascertain an initial state as well as registering an event for when the state changes.

Removing this event may force users to rely on polling the state. Refreshing the browser is a poor UX (and there are implementations of this spec which don't just target browsers!).

Bluetooth on most devices can be turned off, which should change the availability. How would this be handled without the event?

Perhaps the example above should reject the promise when getting availability fails?

beaufortfrancois, 2019-10-25:
Mobile devices provide an easy way for users to turn on/off bluetooth. How do web developers handle this? Does that mean web developers have to listen to a potentially always pending promise to detect when bluetooth is not available anymore with `navigator.bluetooth.getAvailability().then(...)`?

odejesush, 2019-10-25:
The availability is not affected by the powered state of the Bluetooth radio. It's only affected by whether there is a physical Bluetooth radio attached to the system or not [1 & 2]. A site could check availability, then ask the user to insert an adapter and click a button to check again. But this would only really affect devices that have an external adapter, which most devices today have Bluetooth integrated. I hope this clears up any confusion about the API.

[1] [https://webbluetoothcg.github.io/web-bluetooth/#availability-fingerprint](https://webbluetoothcg.github.io/web-bluetooth/#availability-fingerprint)
[2] [https://webbluetoothcg.github.io/web-bluetooth/#availability](https://webbluetoothcg.github.io/web-bluetooth/#availability)

beaufortfrancois, 2019-10-25:
Spec below says "_If the user has configured the UA to return a particular answer from this function for the current origin_". I believe it means user is in control of granting access to the website (think bluetooth permission), therefore this can change during the life of a webpage right?

> The getAvailability() method, when invoked, MUST return a new promise promise and run the following steps in parallel:
> 
> 1. **If the user has configured the UA to return a particular answer from this function for the current origin**, queue a task to resolve promise with the configured answer, and abort these steps.
> 2. If the UA is running on a system that has a Bluetooth radio queue a task to resolve promise with true regardless of the powered state of the Bluetooth radio.
> 3. Otherwise, queue a task to resolve promise with false.

odejesush, 2019-10-25:
You're right, the user can block permissions for a site, and the `availabilitychanged` event can be used to detect when it is available again. But the [Permissions API integration](https://webbluetoothcg.github.io/web-bluetooth/#bluetoothpermissionresult) can also be used by setting the `onchange` handler on the `PermissionStatus`. I think it makes more sense to check the permission using that API.

In the discussion that we had, one of the things considered was only having the `availabilitychanged` event, implementing it such that an event is guaranteed to be fired when you install a listener for it, and removing `getAvailability()`. This would address the weirdness brought up by @yorkie, since developers would only need to use the one API. However, we decided against that because again the potential use of the API seems low, and `getAvailability()` is already shipping in Chrome 78.

I definitely do agree that we should just have one API to reduce the potential bugginess from using the two APIs. Let me know what you think, would it be better to have just one API or both?

yorkie, 2019-10-25:
> In the discussion that we had, one of the things considered was only having the `availabilitychanged` event, implementing it such that an event is guaranteed to be fired when you install a listener for it, and removing `getAvailability()`.

I'd like this solution, actually the `getAvailability()` just acts as a getter for the internal available state, which is also could be served with the `availabilitychanged` event at user-land. Thus removing `getAvailability()` would be better than my initial propose that changing its returned type.

> However, we decided against that because again the potential use of the API seems low,

I think browser should look more like the linux kernel to define its system calls, the high level APIs could be shipped at community libraries, thus the only important to define browser APIs is to provide the abilities correctly, and leverage community for these high-level parts.

> and getAvailability() is already shipping in Chrome 78.

Backwards always is an important item for developers, may a deprecation warning be the only what we can do if we were to change this :(

BTW, just read again about the permission API integration and the followed section "Overall Bluetooth availability", it maybe hard to get the details for WebBluetooth implementors, I also have this kind of following questions,

- What's their relationship?
- Should implementor check the permission before availability returned?
- Should a LSM diagram show the permission & availability state flow?

odejesush, 2019-10-28:
The Permissions API is used to check whether the user has configured the UA to block the use of Web Bluetooth or not, and to query for the list of devices that the current site has permission to use. The availability is mostly for checking whether the current system has a Bluetooth radio or not, but if the UA is configured to block Web Bluetooth, availability is false.

If knowing whether Web Bluetooth is allowed is important to the site, the permission should be checked first before availability. I'm not sure what you mean by a LSM diagram, but a section can be included in the spec to provide more specific information about the relationship between permission and availability.

yorkie, 2019-10-29:
> I'm not sure what you mean by a LSM diagram

That's an algorithm to tell implementor how the permission and availability APIs work.

odejesush, 2019-11-01:
Thanks, I will create a pull request that adds more information about the relationship between availability and permission and an example of how to use the two to determine whether Web Bluetooth can be used or not.