whatwg/bluetooth #462: Why does the Web Bluetooth Pairing Screen Show Multiple Entries for the Same Device?

mduffy215, 2019-11-09:
When my web application goes to pair with my Android device, I get multiple entries for the same device on the pairing screen.

Any ideas why?

[image]

Thx!

mduffy215, 2019-11-09:
Also,  where do the values in the parenthesis come from?

reillyeon, 2019-11-10:
The value in parentheses is the device's MAC address or a placeholder for it. Your device may be rotating its MAC address while advertising. This is a common privacy feature on mobile devices.

This is not a Web Bluetooth specification issue, but a Chromium issue. According to our issue template please file an issue on crbug.com in the "Blink>Bluetooth" component.

closed 2019-11-10 by reillyeon

mduffy215, 2019-11-10:
Thx Reilly.