whatwg/bluetooth #463: Rephrase "triggered by user activation" to match the new user activation model

mustaqahmed, 2019-12-05:
We have User Activation v2 model [in the HTML spec](https://html.spec.whatwg.org/multipage/interaction.html#tracking-user-activation) now.  "Triggered by user activation" should now be changed accordingly.  See the HTML issue mentioned below.

odejesush, 2019-12-05:
Would the user activation for this API (and other device APIs like [WebUSB](https://github.com/WICG/webusb/issues/178)) require a transient user activation?

reillyeon, 2020-01-31:
In Chromium the implementations of both WebUSB and Web Bluetooth require transient activation and do not consume the activation.

mustaqahmed, 2020-02-03:
FYI: One possibility is that WebUSB and WebBluetooth would rely on user activation [indirectly](https://github.com/w3c/permissions/issues/194#issuecomment-564602997) through Permissions API.  Please consider commenting on that Permissions issue, whether the idea sounds reasonable or bad.

PR #470 (2020-02-11): Update language for User Activation v2
This change updates the "request Bluetooth devices" algorithm to refer
to the User Activation v2 model. The algorithm now has to be triggered
while its relevant global object has a transient activation before being
allowed to continue.

This fixes #463.

***
<a href="https://pr-preview.s3.amazonaws.com/odejesush/web-bluetooth/pull/470.html" title="Last updated on Feb 11, 2020, 7:08 PM UTC (2336d55)">Preview</a> | <a href="https://pr-preview.s3.amazonaws.com/WebBluetoothCG/web-bluetooth/470/9991588...odejesush:2336d55.html" title="Last updated on Feb 11, 2 [... cut]
closed 2020-02-11 by odejesush