whatwg/bluetooth #466: startNotifications either gives errors or ties up gatt operations

alexistbell, 2020-01-24:
Chrome 79.0.3945.130
Mac OSX 10.14.6

I'm trying to use Web Bluetooth to control a Bluetooth device and have been running into issues with .startNotifications(). I either get errors back from it, or in the few cases with no errors it seems to tie up the gattOperation until gatt disconnects (or possibly it's causing the disconnection).

I am successfully able to connect to the device, get services, get characteristics, and write characteristics. However some of what I want to do requires listening for events from the Bluetooth device. I’ve tried setting this up in a couple ways.

My first attempt was using async await like this:
```
try {
  await device.gatt.connect();
  const commandService = await device.gatt.getPrimaryService(UUIDS.commandService);
  const commandCharacterstic = await commandService.getCharacteristic(UUIDS.commandCharacteristic);
  await commandCharacteristic.startNotifications();
  await commandCharacteristic.addEventListener('characteristicvaluechanged', handleEvent));   
} catch (e) {
  throw Error(e);
}

const handleEvent = (event) => {
  console.log(event);
};
```

This would give me these errors:
 DOMException: GATT operation failed for unknown reason.
Error: NotSupportedError: GATT operation failed for unknown reason.

When I got to .startNotifications line. If I dont’t try and setup notifications I am able to write to that characteristic just fine, so my connection is there and my UUIDS are correct.

I tried mimic the code [from googles samples](https://googlechrome.github.io/samples/web-bluetooth/notifications.html) more exactly:

device.gatt.getPrimaryService(UUIDS.commandService)
.then((service) => {
  service.getCharacteristic(UUIDS.commandCharacteristic);
})
.then((characteristic) => {
  const myChar = characteristic;
  return myChar.startNotifications()
    .then((_) => {
      console.log('notifications started');
      characteristic.addEventListener('characteristicvaluechanged', handleEvent);
    });
})
.catch((e) => {
  throw Error(e);
});

This gives me the error:

`Cannot read property 'startNotifications' of undefined`

This variation at first seemed the most sucessful:

```
  const commandService = await device.gatt.getPrimaryService(UUIDS.commandService);
  const commandCharacteristic = await commandService.getCharacteristic(UUIDS.commandCharacteristic);

  commandCharacteristic.startNotifications()
    .then(() => {
      console.log('startNotifications');
      commandCharacteristic.addEventListener('characteristicvaluechanged', handleEvent)
        .then(() => console.log(commandCharacteristic));
    })
    .catch((reject) => {
      console.error(reject);
    });
```

But I never got the console log from the .then of `notifications started`. I tried sending my command anyway to see if I would get my response back but got the `Gatt Operation in progress` error which lead me to [this thread](https://github.com/WebBluetoothCG/web-bluetooth/issues/188)

After that I started delaying the commands sent after setting up notifications and ran into some interesting behavior. With a delay of 1000ms I would still get the `Gatt Operation in progress` error.

At 1100ms I would get this error:

N`etworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.`

[MDN](https://developer.mozilla.org/en-US/docs/Web/API/BluetoothRemoteGATTCharacteristic/startNotifications) still has it labeled as experimental, but implies that if I turn on the flag for `#enable-experimental-web-platform-features` I should be able to use it in Chrome. I’ve tried it both with the flag set and unset, and have seen no difference.

Is this feature still too experimental to use? Is there some key piece I’m missing? Is there some way to know when notifications have been successfully started?

I did also find what might be the same sort of issue [here](https://stackoverflow.com/questions/54315828/web-bluetooth-misses-initial-notifications).

odejesush, 2020-01-27:
Hi alexistbell,

As this is an issue with Web Bluetooth's implementation in Chromium and not an issue with the Web Bluetooth specification, please file this in Chromium's bug tracker at https://crbug.com.

closed 2020-01-27 by odejesush

curlyz, 2024-04-26:
For future reference, if you met this error while doing startNotifications →  Uncaught (in promise) DOMException: GATT Error: Not supported.
It is probably because the characteristic you try to subscribe to doesn't have BLE2902 descriptor.

#include <BLE2902.h>
BLE2902* pid2902 = new BLE2902();
charact.addDescriptor(pid2902);

PandaBalls, 2024-07-10:
> For future reference, if you met this error while doing startNotifications → Uncaught (in promise) DOMException: GATT Error: Not supported. It is probably because the characteristic you try to [...]

I added BLE2902, and nRF connect APP on my android phone works well , but in PC Edge browser :
characteristic.startNotifications(); 

then got  DOMException: GATT Error Unknown.

curlyz, 2024-07-10:
> I added BLE2902, and nRF connect APP on my android phone works well , but in PC Edge browser : characteristic.startNotifications();
>
> then got DOMException: GATT Error Unknown.

You also need to add delay between the API calls, I think it has race condition problems internally

PandaBalls, 2024-07-10:
> You also need to add delay between the API calls, I think it has race condition problems internally

`BluetoothDebug[19:03:10] Unexpected GattCommunicationStatus: 3

BluetoothDebug[19:03:08] OnConnectionStatusChanged() status=1

BluetoothDebug[19:03:03] OnGattDiscoveryComplete() success=1

BluetoothDebug[19:03:03] StartGattDiscovery()

BluetoothDebug[19:03:02] No advertisement flags found.

BluetoothEvent[19:03:01] Web Bluetooth Device Chooser initiating Bluetooth discovery session`

Still not works..

TequinDragon, 2025-02-10:
> Hello, I am facing the same problem, any news ?

I've had to deal with this and i've found it's easiest to wrap the `startNotifications` call with a promise and setTimeout to try again if it fails:

```ts
	/** Wrapper around {@link BluetoothRemoteGATTCharacteristic.startNotifications} to try again if the device is busy */
	private async _startNotifications(characteristic: BluetoothRemoteGATTCharacteristic): Promise<void> {
		return new Promise(async (resolve) => {
			let startNotifications = async () => {
				try {
					await characteristic.startNotifications();
					resolve();
				} catch (error) {
					if (!(error instanceof Error) || error.name != 'NotSupportedError')
						throw error;
					console.debug(`Device busy, could not subscribe to characteristic ${characteristic.uuid}. Retrying...`);
					setTimeout(startNotifications, 100);
				}
			};
			await startNotifications();
		});
	}
```