whatwg/bluetooth #479: Allow specify protection/security levels upon requesting a device

tech234a, 2020-03-15:
Some devices require higher levels of security in pairing before accessing their GATT characteristics. For example, when the [battery level example](https://googlechrome.github.io/samples/web-bluetooth/battery-level.html) is tested on an iPhone SE running iOS 13.3 (advertising the battery service using the [nRF Connect](https://apps.apple.com/us/app/nrf-connect/id1054362403) app), and a computer running Chromium 83.0.4086.0 on Windows 10 18363.720, the user receives `SecurityError: GATT operation not authorized.`. On Windows, UWP implements [several different protection levels](https://docs.microsoft.com/en-us/uwp/api/windows.devices.enumeration.devicepairingprotectionlevel) that can be used for Bluetooth pairing, including the `EncryptionAndAuthentication` pairing mode, which presents a code on both devices that the user needs to verify is the same, which appears to be required for accessing many GATT characteristics on iOS.

I believe it would be helpful to have the ability to set a pairing protection level when requesting access to a device with `navigator.bluetooth.requestDevice`. Thoughts?

odejesush, 2020-03-16:
I don't think that `requestDevice()` needs to specify the protection level. The UA should be able to attempt to upgrade the connection to a secure one if it fails to connect to a protected characteristic. This process is different on the various OS'es. macOS performs the pairing process automatically, but other platforms require this logic to be implemented in Chrome.

For Chromium specifically, this [issue](https://bugs.chromium.org/p/chromium/issues/detail?id=960258) provides context on what needs to be done there in order to be able to connect to protected characteristics.

tech234a, 2020-03-16:
Thanks for explaining. It looks like it will be fixed for Chromium 83.

Emill, 2020-03-16:
> I don't think that `requestDevice()` needs to specify the protection level. The UA should be able to attempt to upgrade the connection to a secure one if it fails to connect to a protected [...]

Well I think the idea to "automatically upgrade on failure from peripheral" is broken by design, from a security perspective.

It's very sad that Apple has chosen this idea, since it doesn't give the client side any security promises. The client might of course want to assure a notification comes from a trusted device, or that a write goes to a trusted device. Now a client can't make the requirement that the connection must be MITM protected, for example.

reillyeon, 2020-03-16:
My understanding of the Bluetooth security model is that the peripheral chooses the level of protection supported, not the central. The best we can do to allow a client to enforce a particular level of pairing security is to provide an informative property on the characteristic indicating whether security was required.

Emill, 2020-03-16:
Your understanding is wrong.

Read the Bluetooth Core specification, Vol 3 Part C (GAP), section 10.3.2. The GATT client can have specific security requirements and the standard is extremely detailed how this shall be implemented.