whatwg/bluetooth #482: Getting current device's information

stanleynguyen, 2020-03-23:
It can sometimes be confusing to figure out where the best place is to
file an issue or ask a question about a web platform API. Before
reporting an issue here please ask yourself these questions:

1. Are you reporting a bug in the Web Bluetooth implementation in a
   particular browser? If so then please use their issue tracker. For
   example, the Chromium issue tracker is at https://crbug.com.
N/A

2. Do you have a question about how to use the Web Bluetooth API?
   Please consider asking on StackOverflow and tag it with
   "web-bluetooth". Questions asked on StackOverflow are better indexed
   by search engines and so the answers will help other developers.
   Questions asked here will be answered and the issue closed which
   makes them much more difficult for others to find.

I'm using WebBluetooth API in Chrome 80 with experimental flag enabled. I can retrieved all nearby devices' info with navigator.bluetooth.requestLEScan. However, when it comes to the current device that I'm making scan from, there doesn't seem to be a way to retrieve information about it. Is there anyway that I can retrieve information of the current bluetooth device?

I couldn't find anything from [here](https://developers.google.com/web/updates/2015/07/interact-with-ble-devices-on-the-web), [here](https://googlechrome.github.io/samples/web-bluetooth/), nor [here](https://developer.mozilla.org/en-US/docs/Web/API/Bluetooth)

3. Have you found a problem with the way this API is specified or have
   a feature request? File it here.
If the above mentioned feature has not been implemented yet, this is my request for it to be available.

reillyeon, 2020-03-23:
What information about the Bluetooth central would you like to retreive?

stanleynguyen, 2020-03-24:
@reillyeon  Everything in available `BluetoothDevice` (like info coming for nearby devices). Has it been already implemented

reillyeon, 2020-03-24:
No, it has not been implemented. Please be specific and explain your use case for needing this information.

stanleynguyen, 2020-03-24:
@reillyeon Oh cool cool thanks for the reply.
I'm trying to draw a link between devices, aka which devices have been near to other devices. However, I'm not able to correlate the IDs with devices as I can't get the central device's ID.
There're some forums that suggested using devices' names but I don't think that's a reliable way to go about this

reillyeon, 2020-03-24:
The ability to track which devices have been near other devices is an explicit non-goal of this specification as it would facilitate illicit tracking of the user's location or other private information about the user's environment.

See, https://webbluetoothcg.github.io/web-bluetooth/#bluetooth-device-identifiers and https://webbluetoothcg.github.io/web-bluetooth/#note-device-id-tracking.

stanleynguyen, 2020-03-24:
Thank you @reillyeon

closed 2020-03-24 by stanleynguyen

wscullen, 2023-10-07:
Sorry to open this up again, but I think this needs more discussion. This seems to discount legitimate use cases (positional tracking for which the user explicitly agrees to) on the basis that it COULD be used for nefarious purposes. Should the current device info not be behind a permissions mechanism of some sort instead of not implementing it? Does this remain the situation in October 2023?

reillyeon, 2023-10-09:
So far, allowing tracking of the user's device is out of scope for any of these APIs. A potential API to make the device discoverable (e.g. send advertisements) would also open the door for an API to retrieve information about the central.

wscullen, 2023-10-11:
Ok, I see, thanks for your response.

I don't know enough about bluetooth low energy to ask the right questions but I'll play around with it to see if there is a reliable mechanism for linking devices to users when there is many potential device GUID advertising in a given space, so that a user's device can reliably be used to track indoor positioning, without needing to re-register or calibrate once a user opts in and registers a device for the first time. Should I be looking at UWB for indoor positioning? I would think it's not nearly ubiquitous enough for my use case where the users bring their own devices. Thanks for your time.