whatwg/bluetooth #489: Dead links to developer.bluetooth.org

motla, 2020-04-29:
Hi!
In the [Normative References](https://webbluetoothcg.github.io/web-bluetooth/#normative) section, some links return with error 403 "forbidden":

https://developer.bluetooth.org/gatt/characteristics/Pages/CharacteristicsHome.aspx
https://developer.bluetooth.org/gatt/descriptors/Pages/DescriptorsHomePage.aspx
https://developer.bluetooth.org/gatt/services/Pages/ServicesHome.aspx

odejesush, 2020-04-29:
I'm not sure what these links originally pointed to, but it looks like these are equivalent:
https://www.bluetooth.com/specifications/gatt/characteristics/
https://www.bluetooth.com/specifications/gatt/descriptors/
https://www.bluetooth.com/specifications/gatt/services/

Thanks for pointing out these dead links. I'll put together a PR to fix these as well as some other dead links I can find.

Mausy5043, 2023-02-25:
Those links used to point to tables containing resp. the characteristics, descriptors and services (name/type/number/level). I can also nolonger find those tables.

scheib, 2023-02-27:
Current https://webbluetoothcg.github.io/web-bluetooth/#normative links to Bluetooth SIG appear valid to me.

reillyeon, 2023-02-27:
> Those links used to point to tables containing resp. the characteristics, descriptors and services (name/type/number/level). I can also nolonger find those tables.

We added a version of these files to the registries repo when the Bluetooth SIG removed them from their site:

* https://github.com/WebBluetoothCG/registries/blob/master/gatt_assigned_services.txt
* https://github.com/WebBluetoothCG/registries/blob/master/gatt_assigned_descriptors.txt
* https://github.com/WebBluetoothCG/registries/blob/master/gatt_assigned_characteristics.txt

More discussion of this was on #535.

scheib, 2023-02-27:
The Bluetooth SIG has those values published again at Assigned Numbers URL: https://www.bluetooth.com/specifications/assigned-numbers/, which the https://webbluetoothcg.github.io/web-bluetooth/#normative section links to.