whatwg/bluetooth #494: Web BLE for Flutter

realrk69, 2020-05-04:
I'm not sure this belongs here. But I'll explain briefly.

I've been using Flutter to develop apps for almost 4 months now. Our project has recently received a need to be developed and hosted for the web. The product is an IOT based sensor that operates through BLE.

Question: Is Flutter support for web Bluetooth coming soon? Will it be similar to the javascript promises implementation?

odejesush, 2020-05-04:
This is not the correct repo for this request. This would be one to file in the flutter repo. There seems to be [one that was already filed but closed](https://github.com/flutter/flutter/issues/34848) without a definite response. My suggestion would be to post these questions there.

This issue tracker is only for question about the Web Bluetooth specification itself, not on issues with implementations of Web Bluetooth.

closed 2020-05-04 by odejesush

reillyeon, 2020-05-04:
If Flutter applications can make calls to JavaScript functions provided by the browser then Web Bluetooth is applicable.