whatwg/bluetooth #499: AdvertisingEvent objects may contain service UUIDs that the site has not explicitly requested permission for.

odejesush, 2020-06-01:
Currently, `AdvertisingEvent`s that are fired for `BluetoothDevice`s for which advertisements are being watched may contain service UUIDs that the current site does not have permission to use. This is a privacy risk given that the site is only aware of the services that it explicitly requested for, but it may receive service UUIDs in the `AdvertisingEvent` that is was not aware of. The `AdvertisingEvent` should filter out services that the site did not explicitly request access for.

An example case for when this might happen is the following:
1. A site requests a `device` for service A.
2. The site calls `device.watchAdvertisements()`.
3. The device sends an advertisement packet with services A and B listed.
4. A `AdvertisingEvent` is fired on `device` containing service A and B in the `uuids` property.

jyasskin, 2020-06-01:
Is there a reason the set of service UUIDs is more sensitive than the other things in the [`AdvertisingEvent`](https://webbluetoothcg.github.io/web-bluetooth/#fire-an-advertisementreceived-event), especially the manufacturer data?

reillyeon, 2020-06-01:
I don't think it is particularly more or less sensitive than manufacturer data but the exposure of service UUIDs (and associated data) seems like a hole in the permissions model we have defined for Bluetooth GATT services.

odejesush, 2020-06-01:
Yes, it doesn't align with how we handle service permissions elsewhere. For example, in `getPrimaryServices()`, services that are not allowed are [excluded](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothremotegattserver-getprimaryservices).

Regarding manufacturer and service data, these are also included in the request device filters, but they aren't really used anywhere else besides advertising events. Should these also be filtered out of AdvertisingEvents if they weren't included in the request device options?

reillyeon, 2020-06-01:
I'm not sure manufacturer data is sufficiently structured to make security decisions about, we should treat it like other device properties like `name`. but I think we could apply this to service UUIDs and data.

odejesush, 2020-06-03:
That sounds good to me.

reillyeon, 2020-06-03:
This should likely also apply to the Bluetooth Scanning API and to make that work I think we need to add an `optionalServices` parameter to the [`BluetoothLEScanOptions`](https://webbluetoothcg.github.io/web-bluetooth/scanning#dictdef-bluetoothlescanoptions) dictionary so that sites can express which service's data they are interested in receiving.

odejesush, 2020-06-03:
The `BluetoothLEScanOptions` does include a [`BluetoothLEScanFilterInit`](https://webbluetoothcg.github.io/web-bluetooth/#dictdef-bluetoothlescanfilterinit) which contains a filter for services. The `optionalServices` would be useful if `acceptAllAdvertisements` is true in that case.

This sounds good, I'll work on uploading a commit that introduces these updates.

reillyeon, 2020-06-03:
A new `optionalServices` field is needed because the services in a [`BluetoothLEScanFilterInit`](https://webbluetoothcg.github.io/web-bluetooth/#dictdef-bluetoothlescanfilterinit) are all required for the advertisement to match the filter. Imagine a sensor which alternates between advertising the temperature and its battery level. The temperature service is the one that the application would require in its call to [`requestLEScan()`](https://webbluetoothcg.github.io/web-bluetooth/scanning.html#dom-bluetooth-requestlescan) but could specify the battery service UUID as an optional service so that it could also display that information if the device provides it.

odejesush, 2020-06-24:
As we've discussed offline, we've decided that it would be best to filter the `uuids`, `serviceData` , and `manufacturerData` based on the filters provided to `requestDevice()` when the device was granted. This will make the API consistent with the filtering of not granted UUIDs that is done for other methods, such as `getPrimaryServices()`. Additionally, there will be consistency between service and manufacturer data access. This option operates with intent to allow for the permission UI to be responsive to the type of devices that Web apps seek access to. The solution might be over-engineered or too long-term focused, but it would be hard to retrofit if not implemented.

By default, the name, appearance, TX power, and RSSI are readable without requiring extra permissions. The UUIDs, service data, and manufacturer data are restricted based on their inclusion in `RequestDeviceOptions` when the device was granted. Therefore, the `RequestDeviceOptions` IDL will need to be modified to the following:

```
dictionary RequestDeviceOptions {
  sequence<BluetoothLEScanFilterInit> filters;
  sequence<BluetoothServiceUUID> optionalServices = [];
  sequence<unsigned short> optionalManufacturerData = [];
  boolean acceptAllDevices = false;
};
```

If `optionalManufacturerData` is defined, the manufacturer data that matches the company identifiers included in this sequence will be granted read access for the device that is resolved by `requestDevice()`. All other manufacturer data will be filtered out in  `BluetoothAdvertisingEvent`.