whatwg/bluetooth #500: Robust handling of errors listening to 'characteristicvaluechanged'

rajsite, 2020-06-02:
It's not clear to me what events / behaviors I should listen for to do robust error handling when listening to 'characteristicvaluechanged' events.

For example, I successfully request a bluetooth device, connect to a service, get a characteristic, startNotifications for the characteristic and start listening to 'characteristicvaluechanged' events.

At this point I don't have any pending promises for error states. What are the set of events I need to listen for for all the cases that will result in 'characteristicvaluechanged' failing to result in new values?

Right now 'gattserverdisconnected', 'serviceremoved' for the service associated with the characteristic, and maybe 'servicechanged' (although what if my characteristic was quickly added and remove, is the current event listener still relevant? this is unclear to me)

Are there other events I need to observe? Is it possible to guarantee these are the only ones I need to observe? Could we have an event like 'characteristicvaluechangedunavailable' or something that can be a reliable way to detect when the event listener is no longer valid?

Another way to think about it might be how can I build a robust [ReadableStream](https://developer.mozilla.org/en-US/docs/Web/API/Streams_API/Using_readable_streams#The_ReadableStream_constructor) implementation for characteristic value changes on top of the Web Bluetooth API (because that's what I'm trying to accomplish :P).

Emill, 2020-06-02:
Error handling is done with the `startNotifications()` call. The promise it returns can be rejected if notifications could not be started. Note that this call results in a GATT transaction telling the remote device to enable notifications and the promise result is based on the result of this transaction.

If notifications could successfully be enabled, `characteristicvaluechanged` events will be delivered when notifications arrive. There is nothing that could suddenly cancel these, unless the Bluetooth stack is buggy, so you should never have to worry about this failing. (Until disconnection when notifications stop). UNLESS some other app on the same device writes to the CCCD descriptor to disable notifications (particularly on Android where the Bluetooth stack does not manage the CCCD writing, but the app does, or in this case the Web Bluetooth implementation). Unfortunately the Bluetooth Core GATT/ATT design isn't really built with multiple apps in mind but only sees the GATT client as one single client. All major operating systems multiplex all apps as one single GATT client but a GATT response is only delivered to the app that made the request. So, at least for Android it would be impossible to detect when an other app suddenly tells the remote device to stop sending notifications.

reillyeon, 2020-06-02:
`"gattserverdisconnected"` is the only error you need to consider after the `Promise` returned by `startNotifications()` resolves.

rajsite, 2020-06-02:
> "gattserverdisconnected" is the only error you need to consider after the Promise returned by startNotifications() resolves.

If the service my characteristic is associated is removed then my event listener is garbage. It seems like at the very least I need to listen to both 'serviceremoved' and 'gattserverdisconnected' correct?

reillyeon, 2020-06-02:
The `"serviceremoved"` event is not currently implemented, the section on handling service changes is marked as "unstable".

odejesush, 2020-06-02:
You should be fine if you listen to those events. If you only care about whether the characteristic is still available, the `servicechanged` event is good for that. I think that the `gattserverdisconnected` may not be fired on some platforms if the device goes out of range.

I'm not very familiar with how notifications work. However, quickly glancing through Chrome's implementation of this, if the characteristic is quickly added and removed and the [platform can still produce the same identifier for the characteristic](https://source.chromium.org/chromium/chromium/src/+/master:device/bluetooth/bluetooth_gatt_characteristic.h;l=94), then its possible that it can still listen to notifications from the characteristic.

There isn't really a good way to tell if a characteristic is still notifying that I know of, so perhaps it would be good to have a `BluetoothRemoteGATTCharacteristic.isNotifying` property on the characteristic that can be checked to see if that is still happening. This could be similar to how `BluetoothDevice.watchingAdvertisements` does that for `BluetoothDevice.watchAdvertisements()`.