whatwg/bluetooth #508: navigator.bluetooth.requestDevice filter by deviceId(MAC)

sbahmani, 2020-07-29:
I need to filter the popup selector with deviceId(MAC) because there are many devices with the same name existed and assigned a customized number to each one and persist deviceId with this number in the back-end database. The persisted deviceId not changed for a specific device.
select between many devices with same the name and different deviceId is annoying.

I saw https://github.com/WebBluetoothCG/web-bluetooth/issues/417 but I need filter not direct connect.

reillyeon, 2020-07-29:
I'm going to continue to use #417 to track the request to filter based on device MAC address.

For your use case, does the newly added getDevices() method satisfy your requirements. You can search the list that is returned for a BluetoothDevice with an ID matching your backend database.

See [this guide](https://docs.google.com/document/d/1RF4D-60cQJWR1LoQeLBxxigrxJwYS8nLOE0qWmBF1eo/edit) for now to try out this experimental feature in Chrome.

closed 2020-07-29 by reillyeon

sbahmani, 2020-07-30:
`https://docs.google.com/document/d/1RF4D-60cQJWR1LoQeLBxxigrxJwYS8nLOE0qWmBF1eo/edit` works for those who are already connected. Newcomers need a filter deviceId.

m-khnarges, 2020-08-11:
Hello,

If we does not have filter functionality then we need at least to see MAC addresses, but in iPhone devices (Bluefy app) we only have bluetooth devices names (Below image). It causes users not being able to choose correct device.

[image]

m-khnarges, 2020-08-11:
> I'm going to continue to use #417 to track the request to filter based on device MAC address.
>
> For your use case, does the newly added getDevices() method satisfy your requirements. You can search [...]

Please Check above comment