whatwg/bluetooth #509: BluetoothDevice.watchAdvertisements() not detecting specific devices

TomShorten, 2020-07-29:
I have been working with the watchAdvertisements() function* with BLE medical instruments. The function works completely and allows you to bypass the initial connect screen. However, when I use a TaiDoc instrument* advertisements are never received. 

The initial connect with this device works fine, and is detected immediately (through navigator.bluetooth.requestDevice()). But when trying to use navigator.bluetooth.getDevices() and passing the device into device.watchAdvertisements() it will hang, waiting to receive advertisements.

The same code works for other similar instruments from different company's. But the issue is constant among TaiDoc instruments, thermometer and oximeter.

Could this be an issue with the watchAdvertisements() implementation, i.e. not scanning for all types of advertisements, or is this an issue with the BLE device?

References:
watchAdvertisements() function* - https://docs.google.com/document/d/1RF4D-60cQJWR1LoQeLBxxigrxJwYS8nLOE0qWmBF1eo/edit

TaiDoc instrument* - https://taidoc.com/pulse-oximeter/pulse-oximeter-td-8255/

reillyeon, 2020-07-29:
This looks like an implementation bug rather than a specification issue. Please file such issues against the browser implementing the API (in this case, Chromium) [here](https://bugs.chromium.org/p/chromium/issues/entry) and assign them to the "Blink>Bluetooth" component.

Can you clarify what you mean when you say that watchAdvertisements() will hang? watchAdvertisements() returns a Promise which resolves when the browser has successfully started a Bluetooth scanning session to receive device advertisements. Does the Promise fail or resolve or does your 'advertisementreceived' event handler never get called?

I cannot provide support for particular devices without technical documentation describing their Bluetooth interface. For example, are you use that this device provides data via advertisements and not GATT characteristics?

TomShorten, 2020-07-30:
Thanks for the reply.

I've opened a Chromium case here: https://bugs.chromium.org/p/chromium/issues/detail?id=1111237

In regard to your questions, by watchAdvertisements() will hang, I mean the 'advertisementreceived' event will never fire. The code  looks like this:

device.watchAdvertisements({signal: abortController.signal}).then(() => {
                        console.log('Watching advertisements...');
                        device.addEventListener('advertisementreceived', (evt) => {

When trying to reconnect with the TaiDoc instrument, it will never (or sometimes randomly) pass into the event.

I am unsure whether the device provides data via advertisements and not GATT characteristics, I am in contact with the people who supply the TaiDoc instruments and am also getting them to debug on their end. The advertisement information as shown by the app "LightBlue" can be seen below.
[image]

reillyeon, 2020-07-30:
Thank you for filing that issue. Closing this one.

closed 2020-07-30 by reillyeon