whatwg/bluetooth #513: No services matching UUID help

Cegard, 2020-09-05:
Hello Everyone:

I've been playing around with WebBluetooth since a couple of days ago, for that I've been using my phone with the nRF Connect app as a GATT Server, and I've been using the next function to attempt the connection with the phone:

```
		async function onButtonClick() {
			let device = await navigator.bluetooth.requestDevice({
				acceptAllDevices: true,
				optionalServices: [ 0x180D ]
			});
			let server = await device.gatt.connect();
			console.log(server);
			let service = await server.getPrimaryService(0x180D);
			console.log(service);
		}
```

I've been trying the services: `0x181C`, `health_monitor`, `heart_rate`, but the result I'm getting is: `DOMException: No Services matching UUID 000180-0000-1000-8000-00805f9b34fb`; and with the service `"0000aaa0-0000-1000-8000-aabbccddeeff"` I get `DOMException: Origin is not allowed to access the service. Tip: Add the service UUID to 'optionalServices' in requestDevice() options.`

Any ideas of what could be happening, or another way to test this?

Thanks in advance.

odejesush, 2020-09-05:
To be able to access the services, you need to specify them in the `filters` property of the `RequestDeviceOptions` or in `optionalServices`. In your example, you're using `acceptAllDevices: true`, so you'll need to add the services that you want to access to `optionalServices` in order to get permission to access them.

Cegard, 2020-09-07:
Hello again:

I managed to get to the service (and to the characteristic after that), the solution was adding another service in the `optionalServices` field and then ask for just one of them with `server.getPrimaryService`, something like this:

```
		async function onButtonClick() {
			let device = await navigator.bluetooth.requestDevice({
				acceptAllDevices: true,
				optionalServices: [0x1801, "162348d9-d5a8-4870-8086-8e152fd06a92"]
			});
			let server = await device.gatt.connect();
			console.log(server);
			let service = await server.getPrimaryService("162348d9-d5a8-4870-8086-8e152fd06a92");
			console.log("serviced");
			console.log(service);
		}	
```

If I only specify one of them in `optionalServices` and try to retrieve it, it won't work, is this the expected behavior?

odejesush, 2020-09-07:
Yes, if it's not specified, the site does not receive permission to access
the service. This is intended behavior.

zhangzhoufei, 2020-09-08:
**The same problem as the originator：
error：**
DOMException: Origin is not allowed to access the service. Tip: Add the service UUID to 'optionalServices' in requestDevice() options.

It's been bothering me for a long time
Please help me.
--------------------------------------------
```
navigator.bluetooth.requestDevice({
	 filters: [ 
        { namePrefix: 'play' } 
    ],
    optionalServices: [ "162348d9-d5a8-4870-8086-8e152fd06a92"]
}).then(device => {
	console.log('Got device:', device.name);
	return device.gatt.connect();
})
.then(server => {
	return server.getPrimaryService("162348d9-d5a8-4870-8086-8e152fd06a92");
})
.then(service => {
	
})
.then(characteristic => {

})
.then(value => {
	value = value.buffer ? value : new DataView(value);
	console.log('Battery percentage:', value.getUint8(0));
})
.catch(exception => {
	console.log(exception);
});
```

Cegard, 2020-09-15:
Hey @zhangzhoufei, sorry for the delay, you could try to add another service (besides the one you need to access) in the `optionalServices` field, or deleting that field and use the `getPrimaryServices` function. For me both of these solitions are workarounds, but they worked for me.

dtknowlove, 2020-10-09:
i get the same error,this is my code:
`let device = await navigator.bluetooth.requestDevice({
                filters: [{
                    namePrefix: 'Pai',
                    // services: ['0000fff0-0000-1000-8000-00805f9b34fb','0000fff1-0000-1000-8000-00805f9b34fb','0000fff2-0000-1000-8000-00805f9b34fb','0000fff3-0000-1000-8000-00805f9b34fb'],
                    optionalServices: ['0000fff0-0000-1000-8000-00805f9b34fb']
                }]
            });
            console.log(1);
            console.log('Got device:', device.name);
            console.log('id:', device.id);
            let server = await device.gatt.connect();
            console.log(server);
            await delay(10000);
            console.log(2);
            let services = await server.getPrimaryService('0000fff0-0000-1000-8000-00805f9b34fb')
            console.log(3,services);
            let characteristic = await services.getCharacteristic('0000fff0-0000-1000-8000-00805f9b34fb');
            console.log(characteristic);`

dtknowlove, 2020-10-09:
i get the same error,this is my code:
```
            let device = await navigator.bluetooth.requestDevice({
                filters: [{
                    namePrefix: 'Pai',
                    // services: ['0000fff0-0000-1000-8000-00805f9b34fb','0000fff1-0000-1000-8000-00805f9b34fb','0000fff2-0000-1000-8000-00805f9b34fb','0000fff3-0000-1000-8000-00805f9b34fb'],
                    optionalServices: ['0000fff0-0000-1000-8000-00805f9b34fb']
                }]
            });
            console.log(1);
            console.log('Got device:', device.name);
            console.log('id:', device.id);
            let server = await device.gatt.connect();
            console.log(server);
            await delay(10000);
            console.log(2);
            let services = await server.getPrimaryService('0000fff0-0000-1000-8000-00805f9b34fb')
            console.log(3,services);
            let characteristic = await services.getCharacteristic('0000fff0-0000-1000-8000-00805f9b34fb');
            console.log(characteristic);
```
Any ideas of what could be happening, or another way to fix this?

dvmarinoff, 2020-11-25:
Hi,
I am getting the same error message while trying to get the battery service (0x180F) of a Tacx Heart Rate Belt.
nRF Connect shows the service is there and can read it's Battery Level characteristic, but:
 
- in `chrome://bluetooth-internals/` the service is missing, 
- Also the code example at [web bluetooth samples](https://googlechrome.github.io/samples/web-bluetooth/battery-level-async-await.html) fails with the same error:

`No Services matching UUID 0000180f-0000-1000-8000-00805f9b34fb found in Device.`

Meanwhile I am able to successfully work with the Device Information Service (0x180A), and Heart Rate Service (0x180D).

Here is the log from nRF Connect:

==========================================
nRF Connect, 2020-11-25
Tacx HRB 20483 (F1:17:11:FF:45:F2)
I	14:53:06.050	[Server] Server started
V	14:53:06.058	Heart Rate (0x180D)
- Heart Rate Measurement [N] (0x2A37)
   Client Characteristic Configuration (0x2902)
- Body Sensor Location [R] (0x2A38)
- Heart Rate Control Point [W] (0x2A39)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Model Number String [R] (0x2A24)
Battery Service (0x180F)
V	14:53:06.367	Connecting to F1:17:11:FF:45:F2...
D	14:53:06.367	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	14:53:06.594	[Server callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	14:53:06.594	[Server] Device with address F1:17:11:FF:45:F2 connected
D	14:53:06.630	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	14:53:06.649	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	14:53:06.649	Connected to F1:17:11:FF:45:F2
V	14:53:06.672	Discovering services...
D	14:53:06.672	gatt.discoverServices()
I	14:53:06.835	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	14:53:07.139	[Callback] Services discovered with status: 0
I	14:53:07.139	Services discovered
V	14:53:07.168	Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Heart Rate (0x180D)
- Heart Rate Measurement [N] (0x2A37)
   Client Characteristic Configuration (0x2902)
- Body Sensor Location [R] (0x2A38)
Battery Service (0x180F)
- Battery Level [N R] (0x2A19)
   Client Characteristic Configuration (0x2902)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Model Number String [R] (0x2A24)
- Hardware Revision String [R] (0x2A27)
- Firmware Revision String [R] (0x2A26)
- Software Revision String [R] (0x2A28)
D	14:53:07.168	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	14:53:07.169	gatt.setCharacteristicNotification(00002a37-0000-1000-8000-00805f9b34fb, true)
D	14:53:07.172	gatt.setCharacteristicNotification(00002a19-0000-1000-8000-00805f9b34fb, true)
I	14:53:07.213	Connection parameters updated (interval: 35.0ms, latency: 0, timeout: 5000ms)
I	14:53:12.783	Connection parameters updated (interval: 980.0ms, latency: 0, timeout: 4000ms)
V	14:53:20.830	Reading characteristic 00002a19-0000-1000-8000-00805f9b34fb
D	14:53:20.830	gatt.readCharacteristic(00002a19-0000-1000-8000-00805f9b34fb)
I	14:53:22.579	Read Response received from 00002a19-0000-1000-8000-00805f9b34fb, value: (0x) 64, "d"
A	14:53:22.579	"100%" received
==========================================

reillyeon, 2020-11-30:
@dvmarinoff, please file an issue on crbug.com for this problem and tag it with the Blink>Bluetooth complement. If you can please test on a macOS or Windows device as well.

dvmarinoff, 2020-12-03:
added Issue [1155044](https://bugs.chromium.org/p/chromium/issues/detail?id=1155044)

milesfrain, 2020-12-12:
I'm observing that chrome://bluetooth-internals only lists 8 of 10 available services. The missing services are battery_service `0x180f` and generic_access `0x1800`.
Both of these services appear on nRF Connect and via BlueZ CLI.
```
sudo gatttool -b 84:2E:14:31:BB:40 -I
[84:2E:14:31:BB:40][LE]> connect
Attempting to connect to 84:2E:14:31:BB:40
Connection successful
[84:2E:14:31:BB:40][LE]> primary
attr handle: 0x0001, end grp handle: 0x0008 uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle: 0x0009, end grp handle: 0x000d uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle: 0x000e, end grp handle: 0x001a uuid: 0000180a-0000-1000-8000-00805f9b34fb
attr handle: 0x001b, end grp handle: 0x001e uuid: 0000180f-0000-1000-8000-00805f9b34fb
attr handle: 0x001f, end grp handle: 0x0021 uuid: ec61a454-ed00-a5e8-b8f9-de9ec026ec51
attr handle: 0x0022, end grp handle: 0x002b uuid: 00001815-0000-1000-8000-00805f9b34fb
attr handle: 0x002c, end grp handle: 0x0034 uuid: f598dbc5-2f00-4ec5-9936-b3d1aa4f957f
attr handle: 0x0035, end grp handle: 0x0037 uuid: 1d14d6ee-fd63-4fa1-bfa4-8f47b42119f0
attr handle: 0x0038, end grp handle: 0x0041 uuid: a4e649f4-4be5-11e5-885d-feff819cdc9f
attr handle: 0x0042, end grp handle: 0xffff uuid: 0000181a-0000-1000-8000-00805f9b34fb
```

[image]

Version Info:
```
lsb_release -d
Description:    Ubuntu 20.04.1 LTS
bluetoothd -v
5.53
```

milesfrain, 2020-12-13:
Turns out that a default BlueZ plugin was stealing battery_service `0x180f`.
The fix is to disable the plugin with the `-P battery` flag. More details on the solution are described in https://github.com/WebBluetoothCG/ble-test-peripheral-android/issues/95#issuecomment-743930669

Now chrome successfully detects battery_service `0x180f` :tada:
[image]

travisghansen, 2024-04-15:
I ran into this issue when developing against a custom BLE service (service is running on another linux node). After *adding* characteristics to the service it seems that triggered this issue.

To resolve I have consistently been able to do the following:

```
rm -rf /var/lib/bluetooth/*
systemctl restart bluetooth.service
```