whatwg/bluetooth #517: Does BluetoothDataFilter refer to the data section or raw bytes of an entire Advertisement ?

usivagna, 2020-09-14:
Does the "data" filter in the code snippet below refer to the data section of the advertisement or the raw bytes of the entire advertisement? 
` interface BluetoothDataFilter { 
   constructor(optional BluetoothDataFilterInit init); 
   readonly attribute ArrayBuffer dataPrefix; 
   readonly attribute ArrayBuffer mask; 
};`

If it is the raw bytes, there may be an issue implementing on operating systems that don't provide the entire advertisement.

reillyeon, 2020-09-14:
The data filter applies to the particular context in which it is used, for example service or manufacturer data, and not the entire advertisement.

usivagna, 2020-09-17:
Thanks Reilly, that answers my question. There are a couple other concerns my team has around scanning and I'll open corresponding issues shortly.

closed 2020-09-17 by reillyeon