whatwg/bluetooth #518: Error while writing characteristic to BLE device

pix1289, 2020-09-18:
I need to write a encrypted value to a characteristic of a BLE device using Web Bluetooth API. I can log the service and the characteristic. But I am unable to write the value.

We are referring a python script which can successfully write to the characteristic. The data written is encrypted and then converted to byte array and written to the characteristic.

We are trying to write the same value via javascript to the characteristic but we are getting NotSupportedError. 

Sample code: 

`
navigator.bluetooth.requestDevice ({ 
    filters: [{ name: deviceName }],
    optionalServices: [GATT_SERVICE],
}).then((device) => {
    return device.gatt.connect();
}).then((server) => {
    return server.getPrimaryService(GATT_SERVICE);
}).then((service) => {
    return service.getCharacteristic(GATT_CHARACTERISTIC);
}).then((characteristic) => {
    writeBuffer(data, 0);
    function writeBuffer(data, start) {
        writeOut(data, start);
    }
    function writeOut(data, start) {
            if (start >= data.length) return;
            characteristic
              .writeValue(
                new TextEncoder().encode(data.substring(start, start + 20))
              )
              .then(() => {
                writeOut(data, start + 20);
              });
          }
})
.catch((error) => {
          console.log(error);
        })
`

Since the length is more than 512 bytes, I am taking a substring and trying to write the value. 

What could be the reason this is not working?

odejesush, 2020-09-18:
You've already posted this question on [StackOverflow](https://stackoverflow.com/questions/63957337/writing-characteritstic-value-using-web-bluetooth-api). This issue tracker is for issues about the Web Bluetooth specification, so I'm going to close this issue and try to help you on StackOverflow.

closed 2020-09-18 by odejesush

pix1289, 2020-09-18:
Hello,

Thank you.

Smart-XiaoLeiGe, 2021-01-14:
I think when use writeValue to write more than 33 bytes, will have the issue.