whatwg/bluetooth #528: Update Connection Interval

renu285, 2020-10-17:
Hello All,

Is there any way where the existing connection interval can be updated from an API?

reillyeon, 2020-10-22:
In my research the connection interval cannot be set by the central in the Android or iOS APIs for Bluetooth. Can you point at the API on either of those platforms you would like to access?

renu285, 2020-10-27:
Unfortunately, I am also not aware of such APIs on Android or iOS, I have seen them only in embedded devices where such parameters can be controlled. I was just curious if this was possible in this library.

reillyeon, 2020-10-27:
This specification (for now) only defines an API for devices in the central role. It seems like this ability to control the connection interval is only available to devices in the peripheral role.

closed 2020-10-27 by reillyeon

Emill, 2020-10-27:
This API https://developer.android.com/reference/android/bluetooth/BluetoothGatt#requestConnectionPriority(int) updates connection parameters on Android (for a GATT client). However you only have a few hardcoded values to select among, and it's not even documented exactly which values will be applied for the different configurations.

TeraScience, 2025-02-27:
Hi, I believe the Central device can start a connection parameter update. See the specification bellow:

BLUETOOTH CORE SPECIFICATION Version 5.1 | Vol 6, Part B - 5.1.1  Connection Update procedure

If the master, the slave, or both do not support the Connection Parameters

Request Procedure, then the master shall send an

LL_CONNECTION_UPDATE_IND PDU (the slave shall not send this PDU)

while the slave shall use the L2CAP LE Signaling channel (see [Vol 3] Part A,

Section 4.20 and Section 4.21).

If a device supports the Connection Parameters Request Procedure but does

not know whether its peer does (because neither device has previously

attempted that procedure or performed a Feature Exchange Procedure), then it

shall initiate the Connection Parameters Request Procedure and, if the peer

responds with an LL_UNKNOWN_RSP PDU, by then using the method

described in the previous  paragraph.

dlech, 2025-02-27:
> , I believe the Central device can start a connection parameter update.

Only if you have low-level access to the Bluetooth adapter.  Nothing has really changed in the last 5 years and Android is still the only OS that exposes an API that can even have an effect on this.