whatwg/bluetooth #529: Block scanning of COVID19 codes?

flatsiedatsie, 2020-11-02:
I have used the web bluetooth framework to create something dubious: https://www.coronadetective.eu

Perhaps it would be better if users can't scan for these sensitive signals in the first place?

Emill, 2020-11-02:
I don't really understand this. Am I right that you want to disallow BLE scanning?

flatsiedatsie, 2020-11-02:
Only for the contact tracing codes ("0xfd6f"). There are some legal questions about whether the data that the covid19 contact tracing apps emit could be classified as personal data. As my project shows, if these codes are combined with someone's identity, then they can reveal if someone got covid.

On top of that, in the Netherlands it's technically illegal to process these codes under the temporary "corona law" that went into effect last month.

Google's own Exposure Notification framework doesn't allow other apps to figure out which codes are being transmitted. When why allow websites to pick up these codes from others so easily?