whatwg/bluetooth #530: Correlating Active Scans to location permission [Windows]

usivagna, 2020-11-17:
The ambient advertisements in a user’s area are unlikely to directly include GPS coordinates, but are likely to contain unique identifiers that could be manually correlated with particular physical locations or with particular other people. Given that, the user needs to give  (location related?) permission before a website gets access to nearby advertisements. 

If a user has already given a site permission to know their location, it might be ok to implicitly grant access to BLE advertisements. However, BLE advertisements give away strictly less location information than full [geolocation-api] access, so UAs should allow users to grant that intermediate level of access. 

How does the Web-Bluetooth spec address these location-based permission concerns?

reillyeon, 2020-11-17:
Web Bluetooth addresses this concern by requiring the user grant a site permission to access either a particular device, or advertisements in general (in the separate [scanning](https://webbluetoothcg.github.io/web-bluetooth/scanning) spec).

usivagna, 2020-11-17:
Thanks, I'll be adding a couple other issues based on internal feedback. with our team.

closed 2020-11-17 by usivagna