whatwg/bluetooth #532: NotSupportedError: GATT Error unknown when writing to device using web bluetooth API

pix1289, 2020-11-18:
I am using Angular version 7 on windows 10 and @types/web-bluetooth package.  

My App was working as expected. The data was being written to the device using writeValueWithoutResponse/writeValueWithResponse methods of web bluetooth API. Now it is not working and throwing an error. I have attached chrome logs as well as console logs. Please help. 

PFB my code:
navigator.bluetooth
      .requestDevice({
        filters: [{ name: this.deviceName }],
        optionalServices: [this.GATT_SERVICE], // required if filtering is by name
      })
      .then((device) => {
        // Step 2: Connect to it
        console.log("device:", device);
        return device.gatt.connect();
      })
      .then((server) => {
        // Step 3: Get the Service
        console.log("server: ", server);
        return server.getPrimaryService(this.GATT_SERVICE);
      })
      .then((service) => {
        // Step 4: get the Characteristic
        console.log("service: ", service);
        return service.getCharacteristic(this.GATT_CHARACTERISTIC);
        // return service.getCharacteristics();
      })
      .then((characteristic) => {
        characteristic.writeValueWithResponse(this.data);
      }).catch((err) => console.log(err))
[image]
[chrome logs.txt](https://github.com/WebBluetoothCG/web-bluetooth/files/5558558/chrome.logs.txt)

dlech, 2020-11-18:
I ran into something like this around the Windows 10 2004 update. It turned out that Windows changed how devices were queried and started using "find by type value request" instead of "read by type request" (or something like that). The BLE device didn't handle this new request properly which "Unknown" errors after a timeout.

pix1289, 2020-11-19:
@dlech how did you overcome the issue?

dlech, 2020-11-19:
I followed the instructions to log Bluetooth traffic at https://www.chromium.org/developers/how-tos/file-web-bluetooth-bugs to confirm the bug in the first place. Then I implemented the missing command on the Bluetooth device to fix the failure on Windows.

PandaBalls, 2024-07-10:
> I followed the instructions to log Bluetooth traffic at https://www.chromium.org/developers/how-tos/file-web-bluetooth-bugs to confirm the bug in the first place. Then I implemented the missing command on the Bluetooth device to fix the failure on Windows.

is the missing command is BLE2902()?