whatwg/bluetooth #535: Service and Characteric aliases not obvious (or missing)

milesfrain, 2020-12-14:
_Let me know if https://crbug.com is a better place to post this issue._

When attempting to use a human-readable alias for a service or characteristic, it's not obvious what strings are acceptable.

You can often guess the correct name, but not always. For example:
- `0x180D` "Heart Rate" is aliased as `heart_rate` (as expected)
- `0x180F` "Battery" is aliased as `battery_service` (it seems like this should just be `battery` instead to follow the convention of the other service aliases that all omit the `_service` suffix)
- `0x183B` "Binary Sensor" has no alias (expecting `binary_sensor`)

Error messages point to the spec (https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf)
```
TypeError: Failed to execute 'requestDevice' on 'Bluetooth': Invalid Service name: 'battery'. It must be a valid UUID alias (e.g. 0x1234), UUID (lowercase hex characters e.g. '00001234-0000-1000-8000-00805f9b34fb'), or recognized standard name from https://www.bluetooth.com/specifications/gatt/services e.g. 'alert_notification'.
```
but it would likely be more helpful for developers if the error also pointed to the actual alias mapping https://source.chromium.org/chromium/chromium/src/+/master:third_party/blink/renderer/modules/bluetooth/bluetooth_uuid.cc

Proposing the following changes:
- Add any missing aliases for services and characteristics (e.g. `binary_sensor`, `electric_current`).
  - (note to self) - Also add to [tests](https://source.chromium.org/chromium/chromium/src/+/master:third_party/blink/renderer/modules/bluetooth/testing/clusterfuzz/gatt_aliases.py).
- Add a redundant "Battery" service alias as `battery`.
  - Perhaps add a deprecation warning for the existing `battery_service`
- Modify the [UUID error message](https://source.chromium.org/chromium/chromium/src/+/master:third_party/blink/renderer/modules/bluetooth/bluetooth_uuid.cc;l=367;drc=7eae654e266637310685ce36ec714269ab8d16a8) to also point to the aliases defined in [bluetooth_uuid.cc](https://source.chromium.org/chromium/chromium/src/+/master:third_party/blink/renderer/modules/bluetooth/bluetooth_uuid.cc).

I'm happy to contribute these changes, but wanted to get the green light before diving into building Chromium locally.

milesfrain, 2020-12-16:
Also noticed that the [spec](https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf) is missing some characteristics that are listed in chromium, such as [`0x2A56` 'digital'](https://source.chromium.org/chromium/chromium/src/+/master:third_party/blink/renderer/modules/bluetooth/bluetooth_uuid.cc;l=159;drc=7eae654e266637310685ce36ec714269ab8d16a8).

reillyeon, 2020-12-16:
The source file in the Chromium repository should not be the canonical source for the mapping from names to UUIDs. They should come from either the Bluetooth specifications or the Web Bluetooth specification.

Comments in that file point to pages on bluetooth.com which appear to be deprecated in favor of "16-bit UUID Numbers Document.pdf". This is unfortunate because the older documents provided clear names for each service and characteristic in a form like "org.bluetooth.characteristic.bond_management_feature" which we could use in the mapping. The new documents don't provide similarly useful strings that I would trust to remain consistent long term. Clearly even the old string format wasn't as long terms as we were hoping it would be.

reillyeon, 2020-12-16:
@bluetooth-mdw, do you know what's going on here from a Bluetooth SIG perspective?

bluetooth-mdw, 2020-12-17:
@reillyeon The XML specifications of characteristics and descriptors from which names like "org....." were obtained are no longer used by the Bluetooth SIG. Instead, you should find definitions in the GATT Specification Supplement (GSS) from here: https://www.bluetooth.com/specifications/gatt/ 

That said, I know there are some gaps in that document and that this is being addressed.

reillyeon, 2020-12-18:
@bluetooth-mdw The important thing missing from the new documents are Javascript-friendly names for each of the standardized characteristic and service UUIDs. Does the Bluetooth SIG intend to standardize human-readable names for these or should we deprecate this mapping for the Web Bluetooth API?

bluetooth-mdw, 2020-12-18:
All characteristics and services have a standard name, stated in the corresponding specification. The UUID is a type identifier which will uniquely map to that name. See https://www.bluetooth.com/specifications/gatt/ 

I have a feeling I may be misunderstanding the issue though?

reillyeon, 2020-12-18:
If you view https://www.bluetooth.com/specifications/gatt/characteristics/ through the Wayback Machine you can see what were called the "uniform type identifiers" for each characteristic:

https://web.archive.org/web/20190427040146/https://www.bluetooth.com/specifications/gatt/characteristics/

They seemed to have significance outside of the XML tables. These are what are referred to by the [Standardized UUIDs](https://webbluetoothcg.github.io/web-bluetooth/#standardized-uuids) section of the Web Bluetooth specification.

In the new "16-bit UUID Numbers" document the table only lists a human readable name in the "Allocated For" column and this string seems less canonical than the entries in the old table. Can we assume that these names are standardized and stable? Can you help me understand what it means for a specification to be "withdrawn"?

bluetooth-mdw, 2020-12-18:
This is not really my field. Can you post your questions at https://support.bluetooth.com/hc/en-us please? If possible, log into your company's Bluetooth SIG member account first. If you can't, drill down into any KB article and then you'll find a "Submit a Question" link. Select the best seeming category. If I see it I will assign to our Specification Team as they are the people who will know the history to the uniform type identifier field. 

On "Withdrawal", see https://support.bluetooth.com/hc/en-us/categories/360004202331-Qualifications-and-Listings for FAQ on the topic of Deprecation and Withdrawal. If your question is not answered there, post a separate support request and our Qualification team will answer.

beaufortfrancois, 2021-08-12:
It seems like the list of Bluetooth 16-bit UUID Numbers is available in a PDF format at https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf. It is linked from https://www.bluetooth.com/specifications/assigned-numbers/

After converting to txt, I was able to extract bluetooth services, characteristics, and descriptors with a custom script (https://gist.github.com/beaufortfrancois/1323816074f7383cfa574811abd6ea9c). I've uploaded those changes at https://chromium-review.googlesource.com/c/chromium/src/+/3089126 for those interested.

If we want to continue updating this list. We'd have to come up with rules about naming and policies on UUIDs removal (there are plenty there now).

reillyeon, 2021-08-12:
Since this list is web-exposed we would need to treat removal of an entry as a deprecation. That makes me feel like a mapping should never be removed as existing content may depend on it.

My instinct is that providing this mapping, while increasing readability for developers, was probably a mistake given the long-term maintenance burden it introduces and the lack of clear support from the Bluetooth SIG. I think we should deprecate it and add clear warnings that while the existing entries will not be removed, no new entries will be added and developers should use literal UUID constants instead.

beaufortfrancois, 2021-08-12:
> I think we should deprecate it and add clear warnings that while the existing entries will not be removed, no new entries will be added and developers should use literal UUID constants instead.

I agree. I can work on a Chromium CL if that's fine with you.

beaufortfrancois, 2021-08-12:
Shall we update spec as well?

beaufortfrancois, 2021-08-13:
I've started a Chrome CL at https://chromium-review.googlesource.com/c/chromium/src/+/3093985.

beaufortfrancois, 2021-08-17:
I've just sent an Intent to Deprecate to blink-dev@. See https://groups.google.com/a/chromium.org/g/blink-dev/c/ibDHfEYWOCk

closed 2021-09-02 by beaufortfrancois