whatwg/bluetooth #538: Freezing Chrome Browser on Android device while using scanning functions

jwitas, 2021-02-08:
Hello,

We are trying to use web bluetooth implementation on Samsung Tablet S6 Lite (Google Chrome browser), and we faced an issue when we are using function watchAdvertisement ( or using scanning function from chrome://bluetooth-internals/#devices ) for more then 2 minutes, chrome browser become slower and slower and after 10 and maybe 15 minutes it crashes. (It could be checked by clicking 3 dots to open Settings Menu, it takes sometimes from 10 to even one minute when. this functionality is running for couple of minutes).

Do you know maybe what could be the problem? On PC this problem doesn't exist, so it could be related to android implementation of chrome browser?

Thanks for you support

dlech, 2021-02-08:
Please see https://www.chromium.org/developers/how-tos/file-web-bluetooth-bugs (this repository is just for the specification, not the implementation).

reillyeon, 2021-02-09:
Closing this issue. Please link the crbug.com issue when it is filed.

closed 2021-02-09 by reillyeon