whatwg/bluetooth #553: Get device Id in Notifications and Filter devices by Id

davidwiki8, 2021-07-09:
Is it possible, or will it be, to get the Id of the device sending a notification? Right now, you can only get the UUID of the service and characteristic sending it, along with the data.

In addition, it would be really nice to be able to filter devices acording to their Id, and not only name and services.

reillyeon, 2021-07-09:
Inside a `'characteristicvaluechanged'` event listener you can access the device ID through the `BluetoothRemoteGATTCharacteristic` object like this,

```js
characteristic.addEventListener('characteristicvaluechanged', (e) => {
  console.log(`Device ID: ${e.target.service.device.id}`);
});
```

It doesn't make sense to filter devices based on their ID because the ID is only assigned to a device once the page has access to it. #417 tracks allowing filtering based on MAC address.

davidwiki8, 2021-07-09:
Thanks for the info!

closed 2021-07-09 by davidwiki8