whatwg/bluetooth #555: Bluetooth 4.0 - Secure Simple Pairing (SSP) / Pin-Code Support

mtom55, 2021-07-29:
Currently the web-bluetooth specification does not support submitting a pin code to the device.  Pin codes are quite important in retail locations to stop the public connecting to bluetooth devices.  Additionally some manufacturers do not provide the ability to disable pin-codes.

I think this is called Bluetooth v4.0's Secure Simple Pairing (SSP)

reillyeon, 2021-07-29:
Pairing is currently handled implicitly when a site attempts to read or write to a secure characteristic. #554 recently added notes on support for this in Chromium across supported platforms. We are currently working on making this available consistently.

What are your thoughts on implicit pairing (the current behavior on Android, macOS and soon other platforms) vs. allowing the site to explicitly pair using a programatically provided pin code.

mtom55, 2021-07-30:
I would prefer explicitly programmed pin-codes as it means if you are doing a large rollout of devices there's less steps to that involve the user.  At the very least we would not want the user to have to enter a pin-code more than once.

For example I'd like to pair with 3 devices (2 thermal receipt printers and a label printer in a retail setting).  Within the PWA we could automatically set the pin-code in UI for each device with the option of letting the user change them.

Retail environments are complex because they involve multiple devices and less tech savvy users, so the less you have to get them to do / understand the better.

reillyeon, 2021-08-02:
Thanks for the feedback. Surveying the API landscape it looks like programmatically specifying the PIN code is possible on Android, Linux (BlueZ) and Windows. It does not appear to be possible on macOS. That does seem like enough to make it worthwhile but it will be something we can't guarantee is available for developers.

Emill, 2021-08-02:
The passkey pairing in Bluetooth usually does not make much sense from a security point of view if it's not done interactively by the user. In particular a hardcoded passkey should be avoided if you want to avoid what might seem like unexpected attacks. See https://github.com/zephyrproject-rtos/zephyr/issues/36005.