whatwg/bluetooth #557: Navigator prioritizes one bluetooth device over another one

davidwiki8, 2021-08-13:
I'm currently developing a web app that connects to two motion sensors via the web bluetooth API. They both have the same data output rate. However, when both of them are simultaneosly measuring and sending back the data to their respective event listeners, one of them seems to send almost exactly half the amount of data. Specifically, it's always the device I connect in the first place that sends less data.  It's not an issue with the sensors, as when they are connected alone, they both send the correct amount of data.

Is it posible that the navigator sets a kind of a "priority" on one of them and process more events generated from it because of a shortage of resources?

Thanks in advance!

reillyeon, 2021-08-13:
Air time in the Bluetooth RF band is a limited resource and so it is possible that there isn't enough for both devices to transmit at the same time. If there is a problem here it isn't a Web Bluetooth specification issue (this repository) but an implementation issue. Please file an issue report against Chromium using this link: https://crbug.com/new

Please assign the report to the "Blink>Bluetooth" component and include the following additional details:

1. The model of motion sensor and, if possible, a link to where we could purchase one for testing.
2. The platform being used (Windows, macOS, Android, etc.) and whether the issue is reproducible across multiple platforms.
3. The amount of data being transmitted by the device and the frequency at which it is transmitted.
4. A sample of the code you are using to receive the data which demonstrates the issue.

Add a comment here with a link to your issue report when you are done. Thank you.

closed 2021-08-13 by reillyeon

davidwiki8, 2021-08-18:
https://bugs.chromium.org/p/chromium/issues/detail?id=1240953&q=&can=2