whatwg/bluetooth #558: I want to package a browser that supports webble by myself

TsaiYongChuan, 2021-08-17:
Hello, what do I need to do to package an own android browser

reillyeon, 2021-08-17:
This question is off-topic.

The Chromium project supports Web Bluetooth. Any browser based on it will support the API. How to build your own Chromium-based browser is off-topic for this forum.

closed 2021-08-17 by reillyeon

TsaiYongChuan, 2021-08-17:
Yes, but the entire Chromium project is too big, I just want to make a webview compatible with your webble api, I refer to the ios webble project, this is feasible, but I don’t know how to bridge it

Hope you can recommend some projects or methods to tell me, thank you

TsaiYongChuan, 2021-08-17:
> This question is off-topic.
> 
> The Chromium project supports Web Bluetooth. Any browser based on it will support the API. How to build your own Chromium-based browser is off-topic for this forum.

Yes, but the entire Chromium project is too big, I just want to make a webview compatible with your webble api, I refer to the ios webble project, this is feasible, but I don’t know how to bridge it

Hope you can recommend some projects or methods to tell me, thank you

reillyeon, 2021-08-17:
Web Bluetooth is unfortunately not supported in Android WebView. This is tracked as: https://bugs.chromium.org/p/chromium/issues/detail?id=1100993

There is no project I am aware of which adds Web Bluetooth to Android WebView in the same way that WebBLE adds it to WKWebView.

TsaiYongChuan, 2021-08-20:
> Web Bluetooth is unfortunately not supported in Android WebView. This is tracked as: https://bugs.chromium.org/p/chromium/issues/detail?id=1100993
> 
> There is no project I am aware of which adds Web Bluetooth to Android WebView in the same way that WebBLE adds it to WKWebView.

The web ble project of iOS uses Js Bridge to inject JS code for bridging. I would like to ask if it is feasible on Android. Is there any similar project and code for learning, thank you

TsaiYongChuan, 2021-08-20:
> Web Bluetooth is unfortunately not supported in Android WebView. This is tracked as: https://bugs.chromium.org/p/chromium/issues/detail?id=1100993
> 
> There is no project I am aware of which adds Web Bluetooth to Android WebView in the same way that WebBLE adds it to WKWebView.

Chromium I read part of the Web ble code. All his Bluetooth parts use C++. I can’t expand on it, such as OTA, so I want to make a simple browser for web ble communication based on Android, using only ktolin or java, thank you

TsaiYongChuan, 2021-08-20:
> Web Bluetooth is unfortunately not supported in Android WebView. This is tracked as: https://bugs.chromium.org/p/chromium/issues/detail?id=1100993
> 
> There is no project I am aware of which adds Web Bluetooth to Android WebView in the same way that WebBLE adds it to WKWebView.

Because our project ios uses web ble, in order to be compatible with navigator.bluetooth interface API, I cannot define the protocol myself, so I want to know how to use JsBridge technology on Android to be compatible with navigator.bluetooth

reillyeon, 2021-08-20:
Why do you need to package your own browser on Android? The Web Bluetooth API is supported in Chrome, Edge and other Chromium-based browsers on Android.

TsaiYongChuan, 2021-08-20:
> Why do you need to package your own browser on Android? The Web Bluetooth API is supported in Chrome, Edge and other Chromium-based browsers on Android.

Chromium I read part of the Web ble code. All his Bluetooth parts use C++. I can’t expand on it, such as OTA, so I want to make a simple browser for web ble communication based on Android, using only ktolin or java, thank you

The web ble project of iOS uses Js Bridge to inject JS code for bridging. I would like to ask if it is feasible on Android. Is there any similar project and code for learning, thank you

TsaiYongChuan, 2021-08-23:
> Why do you need to package your own browser on Android? The Web Bluetooth API is supported in Chrome, Edge and other Chromium-based browsers on Android.

 

reillyeon Please help me, thank you

reillyeon, 2021-08-23:
I am sorry but I still don't understand the problem you are trying to solve. Why can't your application be run in one of the many browsers on Android that support Web Bluetooth? Why are you trying to package your own web browser?

TsaiYongChuan, 2021-08-24:
> I am sorry but I still don't understand the problem you are trying to solve. Why can't your application be run in one of the many browsers on Android that support Web Bluetooth? Why are you trying to package your own web browser?

Thank you for your reply, I have made breakthrough progress, I need to customize the browser because your framework does not support me to set MTU and automatically connect to Bluetooth

TsaiYongChuan, 2021-09-08:
Hello, I have successfully implemented all the functions on webview http and can be used normally, but when I try to use it on https, I get an error` Failed to execute'requestDevice' on'Bluetooth': Must be handling a user gesture to show a permission request.` What is causing this, I refer to [](https://github.com/WICG/serial/issues/131) but it is not very helpful to me, can you tell me how to solve it, thank you

reillyeon, 2021-09-08:
Are you calling `requestDevice()` with a user gesture (meaning, inside the event handler for a `click` or similar event)?

TsaiYongChuan, 2021-09-09:
> Are you calling `requestDevice()` with a user gesture (meaning, inside the event handler for a `click` or similar event)?

[image]
[image]
Yes, I put it in the click event

beaufortfrancois, 2021-09-09:
Can you run code below successfully in JavaScript devtools console?
```js
await navigator.bluetooth.requestDevice({ acceptAllDevices:true });
// Prompt should appear because user gesture is not needed in devtools.
```

FYI Error comes from https://source.chromium.org/chromium/chromium/src/+/main:third_party/blink/renderer/modules/bluetooth/bluetooth.cc;l=353;drc=3095f0ce2cadab0ffa2d0ae922ad7dffd3eeda30

TsaiYongChuan, 2021-09-10:
> ```js
> await navigator.bluetooth.requestDevice({ acceptAllDevices:true });
> ```

[image]
[image]

Picture 1 is the result of my running on http, Picture 2 is the result of my running on https, Picture 1 can be used normally, Picture 2 I can’t use

I am not using a chrome browser, but referring to the ios webble project, web bluetooth implemented by injecting JS files, and currently using Android webview

There is a reply record above, thank you

achenging, 2021-09-10:
> Can you run code below successfully in JavaScript devtools console?
>
> ```js
> await navigator.bluetooth.requestDevice({ acceptAllDevices:true });
> // Prompt should appear because user gesture is not [...]

I think it that can not call in bluetooth.cc,because inject  javascript which implement from web bluetooth api.but webview also call in bluetooth.cc and not call  inject javascript.

TsaiYongChuan, 2021-09-10:
> > 你能在 JavaScript devtools 控制台中成功运行下面的代码吗？
> > ```js
> > 等待 导航器。蓝牙。requestDevice ( {  acceptAllDevices : true  } ) ; 
> > // 应该出现提示，因为在 devtools 中不需要用户手势。
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 仅供参考 错误来自https://source.chromium.org/chromium/chromium/src/+/main:third_party/blink/renderer/modules/bluetooth/bluetooth.cc;l=353;drc=3095f0ce2cadab0ffa2d0ae922ad7dffd3eeda30
> 
> 我认为它不能在 bluetooth.cc 中调用，因为注入从 web 蓝牙 api.but webview 实现的 javascript 也调用 bluetooth.cc 而不是调用注入 javascript。

I changed requestDevice to requestDevice2, and found that navigator.bluetooth.xxx is not executing the JS code I injected, but calling the Chrome kernel source code, which is normal on http

> I think it that can not call in bluetooth.cc,because inject javascript which implement from web bluetooth api.but webview also call in bluetooth.cc and not call inject javascript.

I changed requestDevice to requestDevice2, and found that navigator.bluetooth.xxx is not executing the JS code I injected, but calling the Chrome kernel source code, which is normal on http

TsaiYongChuan, 2021-09-10:
> I think it that can not call in bluetooth.cc,because inject javascript which implement from web bluetooth api.but webview also call in bluetooth.cc and not call inject javascript.

I changed requestDevice to requestDevice2, and found that navigator.bluetooth.xxx is not executing the JS code I injected, but calling the Chrome kernel source code, which is normal on http

TsaiYongChuan, 2021-09-13:
amazing，I have solved this world problem

beaufortfrancois, 2021-09-13:
> amazing，I have solved this world problem

How did you do that?

TsaiYongChuan, 2021-09-13:
> > amazing，I have solved this world problem
> 
> How did you do that?

> > amazing，I have solved this world problem
> 
> How did you do that?

I modified the code, instead of executing the navigator.bluetooth module, I left my own code module to bypass navigator.bluetooth

rohitsangwan01, 2022-06-29:
> > > amazing，I have solved this world problem
> > 
> > How did you do that?
> 
> 
> 
> > > amazing，I have solved this world problem
> > 
> > How did you do that?
> 
> I modified the code, instead of executing the navigator.bluetooth module, I left my own code module to bypass navigator.bluetooth

Hey Can you share the javascript code you injected in webBrowser , or repo ,am in same situation,.  it would be really tricky to inject javascript with exact methods from scratch

TsaiYongChuan, 2022-06-29:
> 

> Hey Can you share the javascript code you injected in webBrowser , or repo ,am in same situation,. it would be really tricky to inject javascript with exact methods from scratch

I'm sorry I can't make it available to you, as this code is currently available in stable products, as the company is now at a technological edge that may lead to support for competitors

rohitsangwan01, 2022-06-29:
> I'm sorry I can't make it available to you, as this code is currently available in stable products, as the company is now at a technological edge that may lead to support for competitors

ok no worries