whatwg/bluetooth #561: requestDevice()'s services filter cannot found any device

nex314, 2021-09-15:
I want to use services filter to find my device, but it cannot found any device.
I also test battery_service on https://googlechrome.github.io/samples/web-bluetooth/device-info.html,  still cannot found any device.
but prefix name filter can work. 
How can I do to let services filter work?
thanks.

reillyeon, 2021-09-15:
The services and manufacturer data filters require the device to advertise that data. If you visit about://bluetooth-internals and start a scan it will show the services and manufacturer that is being received from each device.

nex314, 2021-09-17:
Thanks for your kindly replied.
I found that the services filter only can filter service which type is 0x03 but cannot filter out service which type is 0x16?
How can I do to filter out service which type is 0x16?
thanks.

reillyeon, 2021-09-18:
I am not sure what you mean by "service type 0x03" and "service type 0x16". Do you mean advertising data types, where 0x03 is "complete list of service UUIDs" and 0x16 is "service data"?

Is the service you are trying to filter on listed in about://bluetooth-internals in the "Services" column when you scan for devices?