whatwg/bluetooth #566: Fix AbortSignal

miketaylr, 2021-11-15:
In https://github.com/WebBluetoothCG/web-bluetooth/commit/c7c0901a633639250e38e45ea3352ec75dbd6ec1, I attempted to unbreak AbortSignal but didn't do a great job (oops).

Anne gave me some private feedback: 
> misses an equal sign at the end
> Also, it seems the spec still uses AbortError unconditionally, the other change that's needed is to use signal's abort reason instead

I'll try to send in a PR today/tomorrow.

reillyeon, 2021-11-15:
Please fix the missing equals sign, but I don't think we want to use the abort reason unless there's a way to encapsulate it in an AbortError. Using the abort reason will mean that the Promises reject with "undefined" by default which is a rather confusing developer experience.

miketaylr, 2021-11-15:
> Please fix the missing equals sign

SGTM.