whatwg/bluetooth #572: Windows 11 Support

mdmontesinos, 2022-01-05:
Does Web-Bluetooth support new Windows 11 in Chrome (or any other browser)?

I'm asking because we're having issues with an application that currently works in Windows 10, but does not in Windows 11. When the bluetooth devices are connected, they instantly disconnect and trigger the "gattserverdisconnected" event every time. It's the exact same deployed application in both cases, and the same bluetooth devices, but the computer using Windows 11 has this error.

Thanks for any information you might provide!

dlech, 2022-01-05:
Please see https://sites.google.com/a/chromium.org/dev/developers/how-tos/file-web-bluetooth-bugs to report bugs in the implementation (this repo is just for the specification).

mdmontesinos, 2022-01-07:
The issue was posted here:

https://bugs.chromium.org/p/chromium/issues/detail?id=1285296

Thanks for the advice!

closed 2022-01-07 by mdmontesinos