whatwg/bluetooth #583: Access device MAC address

Milooooo1, 2022-04-14:
Hi! I am working on an application where I need the mac address of my device so that I can synchronize it with another device (For the people interested, the device I am talking about is an Xsens DOT sensor). 

However from reading [issue 417](https://github.com/WebBluetoothCG/web-bluetooth/issues/417) its clear to me we do not have access to the device its MAC address. My question is if this is able to be added to the Web Bluetooth API? So far I've found out that the Windows Bluetooth API and Android Bluetooth API do allow you to get the device its MAC address where macOS and IOS do not. I've found nothing about chromeOS.

nondebug, 2022-04-14:
I think the Bluetooth device address (and MAC address) cannot be exposed through the API because some platforms do not provide the device address or provide a synthetic identifier that is not useful for re-identification.

In general, web platform APIs should not expose persistent device identifiers that cannot be reset by clearing the browser's local state. See [Mitigating Browser Fingerprinting in Web Specifications](https://www.w3.org/TR/fingerprinting-guidance/#identifying-fingerprinting-surface-and-evaluating-severity). Bluetooth device addresses are typically [permanent identifiers set in hardware](https://www.w3.org/TR/fingerprinting-guidance/#no-permanent). The device address is intended to be globally unique, making it a significant risk for [active fingerprinting](https://www.w3.org/TR/fingerprinting-guidance/#active-0).

To [mitigate fingerprinting](https://www.w3.org/TR/fingerprinting-guidance/#fingerprinting-mitigation-levels-of-success) while still allowing re-identification of previously connected devices, we should develop an opaque, persistent identifier that can be invalidated when local state is cleared. This would also enable us to expose an identifier on platforms where the device address is obscured.

I think this may not be sufficient for your use case. Can you describe more about how you intend to use the identifier for synchronizing devices?

Milooooo1, 2022-04-14:
Thank you for the detailed answer @nondebug. I understand why it's not possible to access such an identifier. The problem is that according to the Xsens documentation specifically calls for the MAC address of the root sensor this can be seen in the full document which you can download [here](https://content.xsens.com/e3t/Btc/ON+113/cpVSR04/MX6WfwtQqxSW8xRgv02rxsnpW8cYYgJ4FVtKdN5wYXSw3q3pBV1-WJV7CgMQRW8QFpWJ2_QN27W5MmSjm5t4LL3N7ZmlnnZ7hx2W5r4r502sHlMYW13SrWR91jBZJW537N8p3rn_2fW6Z3cND6pR98KW37fwrS6HBGc8W6Vv1td9493qfW2MPzh55gp6-wW1P-mTy7tm0VvW34MKCv5kzFdwW7y9CXb5lxBN8V-2JyJ7y0B3hW1f242Q7Mr9PNN8r4jR5q8prFV__y455PGLJwW1x2WlF3zMfKxW1TRnxn16cBWDF2w5fBqX4fJW7vbw1H8fcPGRW7dJLKS8484DlW8cm9Bw4JQnY_W8F25_07dbySCW1DdFSt5ByyFvW2dk9BN4CFkb-W2M48B85c8q2zW5vCvfm9l0z7vW9f_ygW656R4SW4zrpzC2xKXtd3gPK1) or in the screenshot below. 
[image]

nondebug, 2022-04-14:
Ah, so the Xsens sensor needs the actual MAC address. In that case you will need to find a workaround, for instance looking up the device address in chrome://bluetooth-internals on platforms where the address is provided.

KOLANICH, 2022-04-14:
I vote against exposing MAC. If it is strictly needed, add a text field asking user to type/copy from the needed place in settings.

Flying-Toast, 2024-06-20:
Hi! I'd like to continue this discussion. I am implementing [a protocol](https://github.com/Flying-Toast/qiyi_smartcube_protocol) where I need to send my device its own MAC address as part of a "hello" prelude message. If the device doesn't see its own MAC address in the message you send, it will refuse to carry out any other part of the protocol (stupid I know, but true).

As for the fingerprinting argument, I'd say WebUSB exposes [`USBDevice.serialNumber`](https://developer.mozilla.org/en-US/docs/Web/API/USBDevice/serialNumber), which seems equivalent to exposing a Bluetooth MAC.