whatwg/bluetooth #584: About navigator.bluetooth.requestLEScan() function

nex314, 2022-05-04:
How to implement navigator.bluetooth.requestLEScan() function on Windows Chrome?
The web site has mentions that which can work on Windows 10, but I have no response when I use that function.
https://github.com/WebBluetoothCG/web-bluetooth/blob/main/implementation-status.md

thanks

reillyeon, 2022-05-04:
Can you clarify what you mean by "no response"? Check the DevTools console for errors when calling that function. Because it is still experimental you need to enable `chrome://flags/#enable-experimental-web-platform-features`.

nex314, 2022-05-05:
I enable chrome://flags/#enable-experimental-web-platform-features already.
When the web site running on Android Chrome, which will shown permission dialog before start scan, but there is no permission dialog shown on Windows Chrome and DevTools console has no any error message.

chengweih001, 2022-05-06:
The issue could be reproduced. Please refer to https://bugs.chromium.org/p/chromium/issues/detail?id=1323134.