whatwg/bluetooth #586: Is client to client communication (between nearby devices) possible?

7ombie, 2022-05-08:
Is it possible (using WebBluetooth) to implement a webapp that can communicate with another webapp, running on a different (nearby) machine?

For example, could we build a two-player game, with each player running their own client on their own laptop, and have the two clients communicate game state over Bluetooth?

There are probably more compelling usecases, but that was the case that got me thinking about the Bluetooth API.

beaufortfrancois, 2022-05-09:
It is not possible with the current state of Web Bluetooth. This specification allows websites, running in the **Central role** only, to connect to remote GATT Servers over a BLE connection. In other words, it does not allow websites to act as bluetooth peripherals which would be required for two-ways communication between web apps using Bluetooth only.

I'd recommend looking at https://www.npmjs.com/package/@abandonware/bleno.

7ombie, 2022-05-11:
Thanks, @beaufortfrancois. Much appreciated.

Node is not going to work for me personally. I was considering building something for the Web, but will probably just create a native app to be honest.

Thanks again.

closed 2022-05-11 by 7ombie