whatwg/bluetooth #589: No iBeacon data from WatchAdvertisements()

noahbaron, 2022-06-20:
Platform: Mac OS 11.3.1 (Big Sur)
Chrome: Version 102.0.5005.115 (Official Build) (x86_64)

I'm trying to read iBeacon data using a combination of manufacturer's data filtering, requestDevice and WatchAdvertisements() samples. Unfortunately no manufacturer's data is showing up when I select my iBeacon device. Below is a comparison of different methods used. Is this intentional? I only used the official examples here (https://googlechrome.github.io/samples/web-bluetooth/watch-advertisements.html) and here (https://googlechrome.github.io/samples/web-bluetooth/scan.html?allAdvertisements=true), no modifications made. My hope was to avoid using the "experimental" features in Chrome, but maybe that's necessary to get the iBeacon data? (Major, Minor, Tx Power at 1M)

Data returned from my iBeacon device using the official "Watch Advertisements" sample:

Requesting any Bluetooth device...
> Requested null
Watching advertisements from "null"...
Advertisement received.
  Device Name: null
  Device ID: p8xFey/9h/3Uc7ohPrRiGg==
  RSSI: -38
  TX Power: -128
  UUIDs: 

Data returned from same iBeacon device using official "Scanning" sample:

Advertisement received.
  Device Name: null
  Device ID: p8xFey/9h/3Uc7ohPrRiGg==
  RSSI: -47
  TX Power: -128
  UUIDs: 
  Manufacturer Data: 76
    (Hex) 02 15 a4 95 bb 60 c5 b1 4b 44 b5 12 13 70 f0 2d 74 de 03 0d 28 67 0b
    (ASCII) ¤•»`Å±KDµpð-tÞ(g

closed 2022-06-23 by beaufortfrancois

reillyeon, 2022-06-27:
Note that on Android there is an open issue preventing Chrome from receiving advertisements from iBeacon devices because we are telling the operating system we aren't interested in location-related devices as that requires requesting a different permission than non-location-related devices.

guizmo51, 2022-07-25:
> Note that on Android there is an open issue preventing Chrome from receiving advertisements from iBeacon devices because we are telling the operating system we aren't interested in location-related devices as that requires requesting a different permission than non-location-related devices.

Outch, I tried during days and days before reading this answer. 
bluetooth-internal didn't show the ibeacon but in my demo using watchAdvertisement I received a truncated (only 7 or 8 bytes of data) event (with the Apple ID in the data) 
Do you know how we can bypass this permission ?
thanks !

beaufortfrancois, 2022-07-25:
As noted in https://bugs.chromium.org/p/chromium/issues/detail?id=1296054#c7, as of Android 12 Bluetooth permissions have been extended to include the "neverForLocation" flag.

Chrome includes this flag because the Web Bluetooth API is not intended for determining user location as a replacement for the Geolocation API.

You can read more at https://bugs.chromium.org/p/chromium/issues/detail?id=1324831 as well.

There's no way to bypass this **currently**.