whatwg/bluetooth #591: watchAdvertisements() should return a "sentinel" object

reillyeon, 2022-07-19:
The `requestLeScan()` object returns a "sentinel" object that allows the developer to observe the status of the scan operation and stop it. This is important because some platform Bluetooth stacks enforce a limitation on how long a scanning session can be active and so it is important for a site to know whether to expect to receive more `advertisementreceived` events or not so it can keep its UI up to date.

Right now `watchAdvertisements()` just takes an `AbortSignal` which can be used to stop the scan but this was based on a pattern copied from the Screen Wake Lock API which we eventually decided against, instead adding the `WakeLockSentinel` interface. We should make this change to the Web Bluetooth API as well.

Right now `BluetoothLEScan` doesn't provide an event when the scanning session is stopped (only an attribute) but it could be updated to fire a `stop` event.

niklasdahlheimer, 2023-04-24:
Any news on this topic?

After calling `watchAdvertisements()` on a selected device (filter has `keepRepeatedDevices= true`) the `advertisementreceived`-eventListener stops after a few seconds (tested with Chrome 112.0.5615.137 on MacBook Pro 2021). 
Afterwards the `device` object still has `watchingAdvertisements = true` set, but thats obviously wrong. Further calls to `watchAdvertisements()` have no effect. So it's not possible to monitor advertisement parameters of bluetooth devices continuously.

reillyeon, 2023-04-24:
Thanks for pointing out the problem with the `watchingAdvertisements` attribute as well.

We're tracking this on the implementation side as [Chromium issue 1345995](https://bugs.chromium.org/p/chromium/issues/detail?id=1345995) which is blocking marking `watchAdvertisements()` as a stable API.

niklasdahlheimer, 2023-04-24:
Thanks for referencing the issue @reillyeon !

So, is there currently any way to scan continuously or in regular intervals with this API at all?

I didn't received any scan results using requestLeScan() which I assume is caused by [this](https://github.com/WebBluetoothCG/web-bluetooth/issues/584) issue.

reillyeon, 2023-04-24:
Scan frequency will always be limited by what the underlying platform allows but the idea of providing a sentinel object is to make that visible to the application so that it can restart the scan session after an interval.