whatwg/bluetooth #592: How to invalidate characteristics cache given that Service Changed is not implemented yet?

zachmoshe, 2022-08-30:
I look for the supported way to cause Chrome (or any browser) to invalidate the services and characteristic cache it holds on my device (still under development so I play with the services quite frequently).  As far as I understand, the Service Changed characteristic is the bluetooth way to go, but it seems ([here](https://github.com/WebBluetoothCG/web-bluetooth/blob/main/implementation-status.md)) that it's not implemented yet in any of the browsers.

Is there any other supported (and implemented) API to make the browser rediscover services? 
If not - is there any cache folder that I can delete? (I assume that would be browser specific, I'm referring to Chrome here, but I guess any answer will be useful for some of the future readers)

dlech, 2022-08-30:
The browser doesn't cache the services, it is all done in the OS Bluetooth stack, so clearing the cache is OS-specific.

If the services don't actually change at runtime, I would suggest using the Database Hash characteristic (0x2B2A) instead of Services Changed characteristic or using a random Bluetooth address so that the OS cache is avoided altogether.

zachmoshe, 2022-08-31:
Thanks! I wasn't aware of that.  
IIUC I can just set a random value for the Database Hash Characteristic every time, right?