whatwg/bluetooth #593: "current settings object's relevant global object" is not defined

Ms2ger, 2022-09-16:
https://webbluetoothcg.github.io/web-bluetooth/#handling-visibility-change
https://webbluetoothcg.github.io/web-bluetooth/#handling-full-activity-loss

There's two occurrences of "[current settings object](https://html.spec.whatwg.org/multipage/webappapis.html#current-settings-object)'s [relevant global object](https://html.spec.whatwg.org/multipage/webappapis.html#concept-relevant-global)" in the spec, but a settings object doesn't have a "relevant global object", only a "[global object](https://html.spec.whatwg.org/multipage/webappapis.html#concept-settings-object-global)".

Using "[current global object](https://html.spec.whatwg.org/multipage/webappapis.html#current-global-object)" is also an option.

However, I'm not sure how using the "current" *anything* works here.

reillyeon, 2022-09-19:
The intent of this text is to attach these steps to the "loss of visibility" or "loss of full activity" event on any document. It feels like it should be something like,

> When the user agent determines that a document is no longer [fully active](https://html.spec.whatwg.org/multipage/browsers.html#fully-active), it must run these steps:
> 1. Let _document_ be that document.
> 2. ...

The [abort all active watchAdvertisements](https://webbluetoothcg.github.io/web-bluetooth/#abort-all-active-watchadvertisements) algorithm should really be taking the document (or the document's `navigator.bluetooth` instance) as a parameter.

Does that make sense? Can you suggest the proper way to create such an event listener?

Ms2ger, 2022-09-20:
Thanks! The [`<object>` spec](https://html.spec.whatwg.org/multipage/iframe-embed-object.html#attr-object-name) does something similar:

> Whenever one of the following conditions occur:
> ...
> * the element's [node document](https://dom.spec.whatwg.org/#concept-node-document) changes whether it is [fully active](https://html.spec.whatwg.org/multipage/browsers.html#fully-active),
>
> ...
> (do some stuff)

so I'm guessing something like what you quoted would be fine.

CC @annevk

annevk, 2022-09-20:
Yeah, probably. Getting some of this formalized better is a long-standing issue of the HTML standard.

reillyeon, 2022-09-21:
What do you think of this?

> When a [document](https://dom.spec.whatwg.org/#concept-document)'s [visibility state](https://www.w3.org/TR/page-visibility-2/#dom-visibilitystate) changes,
> 1. Let _document_ be the [document](https://dom.spec.whatwg.org/#concept-document).
> 1. Let _global_ be _document_'s [relevant global object](https://html.spec.whatwg.org/multipage/webappapis.html#concept-relevant-global).
> 1. Queue a task on the Bluetooth event source with global and the following steps,
>    1. If _document_’s [visibility state](https://www.w3.org/TR/page-visibility-2/#dom-visibilitystate) is not "visible", then [abort all active watchAdvertisements](https://webbluetoothcg.github.io/web-bluetooth/#abort-all-active-watchadvertisements) operations on _global_.

I'm unsure when I want to use the "document" concept vs. the "Document" interface.

To get the `Bluetooth` instance from a document it seems like I should go through the [associated Navigator](https://html.spec.whatwg.org/multipage/system-state.html#associated-navigator) (given _global_) and define a similar "associated Bluetooth" concept which is used in the getter steps for the "bluetooth" attribute in the Navigator partial interface.

PR #595 (2025-02-10): Fix visibility change and loss of full activity events
This change fixes the steps for handling visibility change and loss of full activity to find the global object for the Document which experienced the event and from there identify the Bluetooth object for which we need to take action.

This introduces the concept of an "associated Bluetooth", similar to an "associated Document" and adds a "Bluetooth task source" which is now used whenever queuing a task.

Fixed #593.

***
<a href="https://pr-preview.s3.amazonaws.com/WebBluetoothCG/web-bluetooth/pull/595.html" title="Last updated on Feb 10, [... cut]
commit 030375d6 (2025-02-10): Fix visibility change and loss of full activity events (#595)

This change fixes the steps for handling visibility change and loss of
full activity to find the global object for the Document which
experienced the event and from there identify the Bluetooth object for
which we need to take action.

This introduces the concept of an "associated Bluetooth", similar to an
"associated Document" and adds a "Bluetooth task source" which is now
used whenever queuing a task.

Fixed #593.
closed 2025-02-10 by reillyeon