whatwg/bluetooth #604: [watchAdvertisements] Support Extended Advertising

larsgk, 2023-04-18:
Legacy advertising (plus Scan response) can only hold a fairly limited amount of data and extended advertising will see more and more use with e.g. multi-purpose devices.  Extended advertising was introduced with BT 5.0 and most devices sold today support it.

Would it be possible to (also) include data from Extended Advertising, where the OS and hardware (controller) supports it in parallel to the current Legacy Advertising + Scan Response data.

I see a reference to Bluetooth 4.2 in the Web Bluetooth spec.  Is there something preventing this to be bumped to 5.0 (or later)?

reillyeon, 2023-04-18:
I'm struggling to find examples of how platform APIs handle receiving Extended Advertising data. If there isn't anything an application needs to do different to receive such data then it should already work in existing Web Bluetooth implementations, though it would be good to update the specification to explicitly mention it.

larsgk, 2023-04-18:
> I'm struggling to find examples of how platform APIs handle receiving Extended Advertising data. If there isn't anything an application needs to do different to receive such data then it should [...]

It's a different type of response than legacy advertising and scan response, so it has to be included/handled explicitly. Also, I did a quick check against a device advertising both legacy and extended (with the same bt name) and only the legacy data came up (using the watch Advertisements demo page)

dlech, 2023-04-18:
Did you test this on all supported operating systems? (Windows, Mac, Android, ChromeOS)

larsgk, 2023-04-18:
> Did you test this on all supported operating systems? (Windows, Mac, Android, ChromeOS)

Good point. I tested on Android 13/Pixel 7 (TQ2A.230305.008.C1) - will check if the others are also missing extended advertising. However, unless the scanning code is made to support it (like you would combine legacy + scan response), it's probably not going to show up.

reillyeon, 2023-04-18:
If you happen to find example code for Windows, macOS or Android showing how to receive extended advertising data that would be helpful because so far I haven't been able to find anything, which is why I'm assuming it should just work.

larsgk, 2023-04-18:
> If you happen to find example code for Windows, macOS or Android showing how to receive extended advertising data that would be helpful because so far I haven't been able to find anything, which is why I'm assuming it should just work.

Is there a good place to discuss this and collect the info? Doing some quick web searches, it seems there is a bit of confusion around it. Also, I know the controller needs to support it.

larsgk, 2023-04-18:
For android, there seem to be some info here: https://stackoverflow.com/questions/74806907/how-to-filter-ble-devices-that-are-advertising-extended-advertising-message

reillyeon, 2023-04-18:
Thanks. That was exactly what I was looking for.

I've filed https://bugs.chromium.org/p/chromium/issues/detail?id=1434454 to track this from an implementation perspective.

Rakhaau, 2023-07-14:
Keren