whatwg/bluetooth #613: Permissions API Extension

outofambit, 2023-09-28:
As part of developing a cross-browser way to automate managing bluetooth permissions (see https://github.com/w3c/webdriver-bidi/pull/523), members of the Browser Tools and Testing Working Group (BTT) agreed that adding this kind of support would be best done in the Web Bluetooth spec itself, which could then be used by WebDriver BiDi in a generalized pattern (that could be used for other APIs that have more complex permission granting UIs).

I'm opening this issue to track this work by my team at @bocoup as we work towards a proposal. Thanks!

thiagowfx, 2023-10-12:
Any updates on this?

outofambit, 2023-10-12:
@thiagowfx not yet but we'll have more to share in 1-2 weeks!

outofambit, 2023-11-22:
@thiagowfx @nondebug I believe that this issue can be closed in favor of https://github.com/WebBluetoothCG/web-bluetooth/issues/616, which has a more detailed + accurate description of the original goal. I'll close this issue now but feel free to re-open if you assess otherwise. Or if work is opened to address permissions outside of the `requestDevice` flow.

closed 2023-11-22 by outofambit