whatwg/bluetooth #617: p2p connection options

jonnytest1, 2023-12-31:
It can sometimes be confusing to figure out where the best place is to
file an issue or ask a question about a web platform API. Before
reporting an issue here please ask yourself these questions:

3. Have you found a problem with the way this API is specified or have
   a feature request? 
   
   it currently seems like another smartphone does not show up on the pair device(or watch device advertisements) popup, (i have them both unpaired) screen
   
   my use case would be to 
      1. pair a bunch of other smartphones
      2. check their rssi from advertisements to estimate distance to find the devices

nondebug, 2024-01-03:
I think this should work.

What are you trying to connect to on the "another smartphone"? If it's not advertising anything then it won't show up in the chooser dialog.

Try using the Nordic nRF Connect app on the "another smartphone" to create an Advertiser, then you should be able to find it in the Web Bluetooth chooser on the first smartphone.

jonnytest1, 2024-01-03:
🤔 then would it be a good idea to add an "advertise yourself" api to the bluetooth spec ?

nondebug, 2024-01-03:
Advertisement support has been requested in #231 and in Chromium https://crbug.com/1043409.

closed 2024-01-03 by nondebug