whatwg/bluetooth #620: Spec vs Chromium implementation.

hamishwillee, 2024-03-08:
I'm updating MDN docs for Web Bluetooth API in https://github.com/mdn/content/pull/32533. 

The current docs do not match the spec. Before I update, can you confirm the status Chrome vs this spec? Specifically:
1. Does [`Bluetooth.requestDevice()`](https://developer.mozilla.org/en-US/docs/Web/API/Bluetooth/requestDevice#parameters) support the parameter options as documented in spec? 
  - The MDN docs currently document filters as `BluetoothScanFilters`, `optionalServices`, and `acceptAllDevices`. I haven't checked the detailed definitions.
  - Do we know the chrome version in which the changes to this new API would have happened? (for browser compatibility data)
2. Permissions API integration - according to docs `query()` takes same options as `Bluetooth.requestDevice()`. Are all the options implemented? If not, which are implemented, and in what versions were they added?

Thanks!

reillyeon, 2024-03-09:
Chrome implements the `Bluetooth.requestDevice()` method, including its `options` parameter. The MDN refers to the `filters` option as an array of `BluetoothScanFilter` but the spec and Chrome use the name  `BluetoothLEScanFilterInit`, not that this difference is exposed to developers.

While the Permissions API integration is implemented in the sense that the algorithms there describe how permissions are managed the integration with the script-exposed `query()` method is not. Developers must use `Bluetooth.requestDevice()` or the upcoming `Bluetooth.getDevices()` methods to request and query permissions. The `getDevices()` method is under development and tracked by [this ChromeStatus entry](https://chromestatus.com/feature/4797798639730688).

hamishwillee, 2024-03-11:
> While the Permissions API integration is implemented in the sense that the algorithms there describe how permissions are managed the integration with the script-exposed query() method is not. 

Thanks @reillyeon - much appreciated.

I understand that you are saying  that you can't specify options to query and get back permitted devices. It sounds like instead users would check the permitted existing devices using `getDevices()`?

Just FMI:
1. Is the intent that you might be able to filter devices using `getDevices()`?
2. Is the plan ever to match the spec w.r.t. returning device, or is the intent to get the spec to match Chrome?

reillyeon, 2024-03-11:
> I understand that you are saying that you can't specify options to query and get back permitted devices. It sounds like instead users would check the permitted existing devices using `getDevices()`?

Correct.

> 1. Is the intent that you might be able to filter devices using `getDevices()`?

Filtering devices isn't supported by `getDevices()`. It's unclear how useful this would be to developers. 

> 2. Is the plan ever to match the spec w.r.t. returning device, or is the intent to get the spec to match Chrome?

Supporting querying Web Bluetooth device permissions with `navigator.permissions.query()` would be nice, but when I looked into supporting it for WebUSB there were some technical limitations in the Chromium implementation which meant it was difficult to implement and so I decided it wasn't worth the effort given that the functionality is duplicated by the `getDevices()` method. 

> 1. are the spec event handlers all implemented, such as [advertisementreceived](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothdeviceeventhandlers-onadvertisementreceived) (these aren't documented on MDN). Do we know the version they appeared in?

The `advertisementreceived` event handler is used by two related features `BluetoothDevice.watchAdvertisements()` and `navigator.bluetooth.requestLeScan()`. Both of these are currently prototyped in Chromium but are not yet enabled by default.

> 2. What form does `optionalManufacturerData` actually take? It looks like an array of numbers, but what do those numbers map to? How are they used? It reads as though this is not used for filtering, but is data you want to make available to the origin after filtering.

It operates similarly to `optionalServices`. It doesn't filter the devices displayed to the user in the permission prompt but controls which manufacturer data fields provided in `advertisementreceived` events. Bluetooth manufacturer IDs are integers. The "optional" versions of these parameters are useful because it allows the developer to declare the data they are interested in receiving from the devices without overly constraining the filter controlling which devices are presented to the user in the permission prompt. 

> 3. The spec seems to indicate that `options` is optional, but not clear what gets fetched in this case. Thoughts?

You're right that `options` is not truly optional because one of `filters` or `acceptAllDevices` is required.

hamishwillee, 2024-03-12:
Thank you very much. I'm going to close this as answered "for now". When the currently experimental bits go live MDN will probably still need quite a bit of work. 

FWIW from a developer point of view it is always best if specs match implementation. If matching the spec as it is is not feasible, then working to modify the spec while there are no other implementers would be a "good thing". I'm sure you know that.

closed 2024-03-12 by hamishwillee