whatwg/bluetooth #623: Proposal: active scanning with scan response

mikolajefento, 2024-04-05:
Currently, there is no option in `requestLEScan` to listen for scan responses, nor is there a field in `BluetoothAdvertisingEvent` to access the scan response. I believe it would be beneficial to introduce 'normal' and 'active' scan modes, as there appears to be confusion in various implementations at present.

For instance, in Chromium, the `watchAdvertisements` method returns two subsequent events: one with standard manufacturer data and another with the scan response. However, when using requestLEScan, only the standard advertising response is provided.

I propose that establishing a clear distinction would, at the very least, aid in determining which API is stable enough in different implementations to effectively utilize scan responses.

dlech, 2024-04-05:
This sounds like platform-specific differences showing through. 

On Windows, different events are received for each advertisement packet received so we get separate events for "regular" advertisement data and a separate event for the scan response. But on macOS, Android and Linux events only indicate that something in the advertisement data changed and there is no way to tell on those platforms which information was in the advertisement data and which data was in the scan response. So this one is probably more of an implementation issue where Windows needs to be made to work more like the other platforms under the hood so that web developers don't have to handle OS-specific differences in their apps.

Also, on macOS, passive scanning is not possible (also not an easy way to do it on Linux). So I think it only makes sense to have active scanning in WebBLE.

alvinjiooo, 2024-04-05:
Just to clarify, does the "scan response" mean `serviceData` in `advertisementreceived`? 
Both `watchAdvertisements` and `requestLEScan`  induce `advertisementreceived` so I assume if scan responses are available to one it should also be available to another?

dlech, 2024-04-05:
"scan response" in this context means an advertisement data packet with a PDU Type of `SCN_RSP`.  What types of advertisement data it contains depends entirely on the peripheral device. 

I would expect `watchAdvertisements` and `requestLEScan` to return the same info for the same device, so if it isn't, that sounds like a potential implementation bug to me.

alvinjiooo, 2024-04-05:
Yes, I would assume `watchAdvertisements` and `requestLEScan` to see the same information if the advertisement event matched what filter defined.  Although the complete filtering implementation is not finished yet. 

In Chromium:
content/browser/bluetooth/advertisement_client.cc
// TODO([crbug.com/707635](https://crbug.com/707635)): Support manufacturerData and serviceData
    // filters.

reillyeon, 2024-05-01:
From the conversation above it sounds like the issue is whether two `advertisementreceived` events are fired, each with data from a single Bluetooth packet or one `advertisementreceived` event is filed with the combined data from both packets. I'm trying to understand whether this difference matters to applications because there might be some data missing or incorrectly combined, or if all the data is present but applications need to consider the possibility it will be split across multiple events. The former would be a serious issue while the latter is annoying but workable.

mikolajefento, 2024-05-02:
@reillyeon, based on my experience with Chromium implementation, I've observed missing data when using `requestLEScan`. This could be due to incomplete implementation, as pointed out earlier. On the other hand, with `watchAdvertisements`, data arrives separately on Windows but combined on Android, which is indeed annoying but workable 😄.

Considering this, it's conceivable that different browsers may handle data combination differently, leading to potential debugging challenges. To streamline this process, a simple addition to the Web Bluetooth spec could be added. If the specification explicitly stated whether scan responses should be combined or split, browser developers could align their implementations accordingly facilitating long-term maintenance.