whatwg/bluetooth #635: Make "simulated Bluetooth adapter" browser-wide

sadym-chromium, 2024-11-08:
Current CDP implementation allows only a browser-wide "simulated Bluetooth adapter", while specification requires it to be browsing-context specific. Switching to navigable-specific will be a breaking change.

We propose the following: make the "context" parameter of "bluetooth.SimulateAdapterParameters", "bluetooth.SimulatePreconnectedPeripheralParameters" and "bluetooth.SimulateAdvertisementParameters" optional. This would allow for both navigable's specific and global "simulated Bluetooth adapter". For now, until CDP implements navigable-specific "navigable's specific "simulated Bluetooth adapter", the implementation will return an error.

Note that probably "simulated Bluetooth adapter" is [user context](https://w3c.github.io/webdriver-bidi/#user-contexts)-specific.

reillyeon, 2024-11-08:
Given the permission model for Bluetooth devices, I think only supporting a browser-wide simulated adapter makes sense. Automation needs to combine the use of the simulated adapter with automation of the Bluetooth permission prompt to grant a given site access to a simulated device.

Thinking about it from this perspective I don't know why I pushed so hard for it to be browsing-context specific.

sadym-chromium, 2024-11-18:
Related issue: https://github.com/WebBluetoothCG/web-bluetooth/issues/637

reillyeon, 2024-11-18:
Thinking about this a bit further in the context of your other issue, I realize that while theoretically we could have simulated Bluetooth devices appear to all pages globally the reality of how this is implemented in Chromium makes it hard to mix real and simulated devices. The Bluetooth simulation commands also support configuring the state of a simulated adapter (e.g. powered off) and if that state were global then it wouldn't allow mixing of real and simulated devices anyways.

With that context, I think the current approach of enabling a simulated Bluetooth adapter for a particular top-level traversable is best.

OrKoN, 2024-11-19:
I too think that the current way the spec is written makes the most sense from test author/client perspective (i.e., configuring the simulated devices per top-level traversable). I think the Chromium implementation should follow that (if possible) but I do not think we need address this mismatch immediately. 

@reillyeon @alexnj  do you know if there is an existing crbug to track aligning the CDP implementation with the specification? 

Although if it is really hard to achieve in the Chromium bluetooth implementation then perhaps we should consider changing the spec instead. 

> Current CDP implementation allows only a browser-wide "simulated Bluetooth adapter", while specification requires it to be browsing-context specific. Switching to navigable-specific will be a breaking change.

I didn't get the "Switching to navigable-specific will be a breaking change." part? Is it a breaking change for the specification? I think it is already specified in terms of top-level traversables.