whatwg/bluetooth #636: BluetoothLEScanFilter's manufacturerData and serviceData are not arrays

jeroen1602, 2024-11-15:
The `BluetoothLEScanFilters` dictionary is created from an `BluetoothLEScanFilterInit` but its `manufacturerData` and `serviceData` fields aren't an array.

https://github.com/WebBluetoothCG/web-bluetooth/blob/f3b31870764ce76fa1fa671f7bcdac7944a246a0/scanning.bs#L374-L382

In the `BluetoothLEScanFilterInit` dictionary that is used in the constructor they are referenced as being a `sequence` so when converting to an `BluetoothLEScanFilters` what needs to happen to the extra items in the array?

https://github.com/WebBluetoothCG/web-bluetooth/blob/f3b31870764ce76fa1fa671f7bcdac7944a246a0/index.bs#L529-L535

The docs seem to assume that the fields are singular and not an array.

https://github.com/WebBluetoothCG/web-bluetooth/blob/f3b31870764ce76fa1fa671f7bcdac7944a246a0/scanning.bs#L419-L426

jeroen1602, 2024-11-15:
Just checked what chromium does. And it seems that it uses the same type for both `requestLEScan` and `requestDevice`

https://github.com/chromium/chromium/blob/203588b13d3fa2a4ad4a463823e4b574ed39c8f7/third_party/blink/renderer/modules/bluetooth/bluetooth_le_scan_options.idl#L7-L11

https://github.com/chromium/chromium/blob/203588b13d3fa2a4ad4a463823e4b574ed39c8f7/third_party/blink/renderer/modules/bluetooth/request_device_options.idl#L7-L13

chengweih001, 2024-12-18:
Thanks for reporting this. like what you pointed out in the Chromium implementation,I suspect [web-bluetooth/scanning.bs](https://github.com/WebBluetoothCG/web-bluetooth/blob/f3b31870764ce76fa1fa671f7bcdac7944a246a0/scanning.bs#L419-L426) should just use `BluetoothLEScanFilterInit` from [web-bluetooth/index.bs](https://github.com/WebBluetoothCG/web-bluetooth/blob/f3b31870764ce76fa1fa671f7bcdac7944a246a0/index.bs#L529-L535). We will look into this and update this when we are actively working on shipping Web Bluetooth Scanning.