whatwg/bluetooth #640: How to properly abort a `device.gatt.connect()` attempt

mgibbs189, 2025-01-23:
Hello,

Could someone please describe the "correct" way to handle a failed reconnection attempt?

When `gattserverdisconnected` fires, I want to try reconnecting for up to 30 seconds, then abort. This seems like a common use-case.

```js
let device, server;
let timeout;

let timeoutPromise = new Promise((_, reject) => {
    timeout = setTimeout(() => reject(new Error('Connection timeout')), 30000);
});

try {
    server = await Promise.race([device.gatt.connect(), timeoutPromise]);
    // successfully reconnected, do other stuff
}
catch(err) {
    // failed to reconnect, how to abort connect()?
    // this often causes `navigator.bluetooth.requestDevice()` to stop responding, perhaps
    // because the browser is still actively trying to connect?
}
finally {
    clearTimeout(timeout);
}
```

It'd be super helpful if someone could clarify what happens with `connect` behind the scenes...

* Is there actually a way to abort `device.gatt.connect()`?
* Is there any browser-level timeout? Or, will a `connect()` run indefinitely (i.e. the lifetime of the pageload)?
* What does `gatt.device.disconnect()` do, when a device is disconnected/dropout, or when a `connect()` is running for that device?

Thanks

dlech, 2025-01-23:
See https://github.com/WebBluetoothCG/web-bluetooth/issues/152

mgibbs189, 2025-01-23:
Thanks @dlech -- yikes, it's almost a decade old.

What do you think are the chances we can get some traction on this issue? I'm not super familiar with Chromium's dev cycle.

This seems like a pretty fundamental issue 😬

reillyeon, 2025-01-24:
I believe the Chrome team is tracking this as a higher-priority issue when combined with persistent permissions as it becomes more likely a site will try to connect to a device that is not in range.

@nondebug

mgibbs189, 2025-01-24:
@reillyeon You're on the Chromium team, right? Mind if I ask your exact involvement?

I'd love to see this one get resolved, and will do everything in my power to help make that happen. It's such a critical feature.

What does the Chromium team need from me and/or the community in order to get the ball rolling on this one?

reillyeon, 2025-01-24:
I am on the Chrome team but am not as involved with Web Bluetooth as I have been in the past.

As much as it feels like shouting into the void please keep posting about the importance of fixing issues like this. It helps if you provide information about the application you are building and the company you work for so that we get a sense of the impact of missing features like this so we can prioritize.

If you're comfortable with C++ of course patches are welcome but our team will try to get to things like this. There are just so many features and so little time.

nondebug, 2025-01-24:
The Chromium bug filed for this issue (https://crbug.com/40502943) is marked as a blocker for [crbug/40632400: Implement persistent Web Bluetooth permissions](https://crbug.com/40632400) which is on our roadmap for 2025.

As @reillyeon said, providing information about how your company will use this feature helps us justify this work against other high-priority work.

mgibbs189, 2025-01-24:
@nondebug @reillyeon please see this post for more background: https://issues.chromium.org/issues/40502943#comment33

Would it be a possibility for Google to offer some sort of sponsorship or grant to fast-track development? I was contacted by a dev agency willing and able to contribute, but I don't personally have the funding to sponsor it.